nodemcu/nodemcu-firmware #1638: Add support scripts to Travis-CI.

karrots, 2016-12-04:
### Missing feature
Running of all support scripts in Travis-CI.

A few examples:
* lua.cross
* spiffs-image

### Justification
See #1637 
### Workarounds
None

marcelstoer, 2016-12-04:
> spiffs-image

It does run during a CI build, see #1502.

karrots, 2016-12-04:
ah ok. I was trying to think of anything else at the same moment. Forgot
its part of the regular build as I run it separately in my build scripts.

TerryE, 2016-12-05:
The underlying point that @karrots makes in #1637 still applies here we don't do a validation build of luac.cross as part of the Travis check, so don't pick up this sort of issue.   

@marcelstoer, am I correct in assuming that runs as a nosudo container?  The issue that we have with luac.cross is that it uses a lua script to build it, which means that we have to lave Lua installed on the system.  This is trivial is we have sudo but a bit more convolved if we want to install a local copy.  As far as I can see there are three options if we can to build luac.cross:
-  Use a sudo container and sudo to pull lua and lua-rocks from the ubuntu repo
-  Develop a (non-standard) make-based build of luac.cross
-  Use some thing like [hererocks](/mpeterv/hererocks) which uses a Python boostrap to but a user version of lua and lua rocks instead of the prebuild Ubuntu repo ones.

karrots, 2016-12-05:
>    - Use some thing like hererocks <http:///mpeterv/hererocks> which uses

Apt works in the container based builds via the APT add-on.

https://docs.travis-ci.com/user/installing-dependencies/#Installing-Packages-with-the-APT-Addon

karrots, 2016-12-05:
That said `lua-filesystem` and `lua5.1` are not in the list of whitelisted packages.

https://github.com/travis-ci/apt-package-whitelist/blob/master/ubuntu-trusty

Assuming the packages can pass the tests and be whitelisted the `luac-travis-ci` branch on my repo should work.

jmattsson, 2016-12-05:
Getting the necessary packages white-listed should be the first avenue of approach. It's both the right thing to do, and the easy thing.

marcelstoer, 2016-12-05:
Both packages are whitelisted for ubuntu-precise which we use (we don't declare the Ubuntu version explicitly).

karrots, 2016-12-05:
For some reason it fails to install them on my branch. All I did was add
three lines to the build file.

marcelstoer, 2016-12-05:
I suggest you contact Travis support. https://github.com/travis-ci/apt-package-whitelist/blob/master/ubuntu-precise#L9926 ff is quite clear and I see nothing obviously wrong with your change.

TerryE, 2016-12-05:
@karrots Jonathan, I use a Virtualbox VM that mirrors the Travis Docker base image.  I can make Vbox dance, but haven't got to grips with Docker.  Anyway I installed `lua5.1` and `luarocks` and then used rocks to install the rocks package `luafilesystem` (and a couple of other packages that I use such as `md5`, `bit32` and `luasocket`).

karrots, 2016-12-05:
I have a slightly modified version of @marcelstoer's Docker build image with ubuntu packages `lua-filesystem` and `lua5.1` installed via `apt-get` which builds lua.cross fine. That's actually how I found the error.  Adding those same packages to the Travis CI file doesn't find those packages. I'll create a ticket with their package team. Once fixed I'll create a PR on this repo to add the build to CI.

TerryE, 2016-12-06:
>  Adding those same packages to the Travis CI file doesn't find those packages. I'll create a ticket with their package team.

@karrots Jonathan, why don't you just follow my suggestion and use `apt-cache search` to report what packages are bundled.  It might be something just as simple as you've got the package name wrong or need to rocks.

karrots, 2016-12-06:
Good point Terry. Looks like Ubuntu 12.04 doesn't contain lua5.1 or lua-filesystem in their repository. Any reason CI can't be upgraded to 14.04? 12.04 runs out of support early next year.

TerryE, 2016-12-06:
Yes AFAIK, Travis now supports Ubuntu 14.04 for **sudo-enabled** environments, but is still using Ubuntu 12.04 for **container-based** ones.  This sort of change request really merits its own issue and discussion.
The current Travis builds are container based, so you need to install any package in your user account and not **root**.

We need to use hererocks that I referenced above to compile lua in a user account, or alternatively we could just create a tarball with the lua exe and lfs in it and unpack it much as we do with the SDK.

karrots, 2016-12-06:
Travis supports both 12.04 and 14.04 in container based builds. Most of our packages would need to be added to the white list. Benefits to apt are the travis-ci file stays a bit cleaner.

I have made requests for the various needed packages to be added to 12.04 and 14.04. In the meantime, I have created a new branch using hererocks. 

It seems to work fine albeit it can't be merged until #1647 PR is merged fixing the issue with lua.cross building.

marcelstoer, 2016-12-06:
> Any reason CI can't be upgraded to 14.04? 12.04 runs out of support early next year.

> Travis now supports Ubuntu 14.04 for sudo-enabled environments, but is still using Ubuntu 12.04 for container-based ones. This sort of change request really merits its own issue and discussion.

Yes, for the moment [container-based environments still use 12.04](https://docs.travis-ci.com/user/ci-environment/#Virtualization-environments), you don't have a choice. I'm sure Travis will take appropriate measures when it reaches EOL.

> alternatively we could just create a tarball with the lua exe and lfs in it and unpack it

👎 not an option for me, for the reasons discussed elsewhere (repo history/size).

TerryE, 2016-12-06:
@marcelstoer Marcel, I don't think that we should close this.  I agree with your comment that we have to use Ubuntto 12.04 with Travis container-based environments, but this fact doesn't close the original issue raised.  It's just a constraint.

We include `luac.cross` and `spiffs-image` in our build so Travis CI should verify that we can build them.   

I also don't see the issue with including another 90 Kb tarball that is going to remain the same whilst we are on the U 12.04 containers.

marcelstoer, 2016-12-06:
I closed it after I merged #1649 since that resolves this issue IMHO.

TerryE, 2016-12-06:
@marcelstoer Marcel, apologies from an old man :)  I missed that Jonathan had implemented my suggested work-around whilst I was sleeping!!  I will reclose this.  

Sorry again.

PR #1649 (2016-12-06): Add lua.cross to CI tests.
Fixes #1638.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/en/*`.

The full build environment was not being tested. Changes were [... cut]
closed 2016-12-06 by TerryE