nodemcu/nodemcu-firmware #1639: wifi.setmode not works from first call and doesn't save configuration

Nikolay-Ch, 2016-12-04:
### Actual behavior
I Reset my module. Through ESPlorerI I send this line:
`="WiFi mode: " .. wifi.getmode()`
I get:
> WiFi mode: 2

Then I send this two lines:
```
wifi.setmode(wifi.STATION, true)
="WiFi mode: " .. wifi.getmode()
```
Again I get:
> WiFi mode: 2

And only when I send this lines:
```
wifi.setmode(wifi.NULLMODE)
wifi.setmode(wifi.STATION, true)
="WiFi mode: " .. wifi.getmode()
```

I get:
> WiFi mode: 1

First question is - why setmode not working in one call?

then I reset my module and send line:
`=wifi.getmode()`

I get:
> WiFi mode: 2

So configuration doesn't saves in to the memory. This is a second question - why?

### NodeMCU version

> NodeMCU custom build by frightanic.com
> 	branch: master
> 	commit: ec265a6c21db22640795f190bdcb8a4f014cdced
> 	SSL: true
> 	modules: cjson,dht,file,gpio,i2c,net,node,rtctime,sntp,tmr,wifi,ws2812
>  build 	built on: 2016-12-03 23:46
>  powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
> 

### Hardware
WeMos D1 Mini Pro

devsaurus, 2016-12-04:
Looks like a duplicate of #1634.

You could try to work around this issue by *not* saving the settings to flash and run a script each time after booting the firmware. This seems to work for my Mini Pro.

Nikolay-Ch, 2016-12-04:
My project works fine. But I think that if saving to memory work properly, it would be better...
In addition to #1634, I find workaround to set mode 1 by calling setmode function twice - first with mode 0, then with mode 1.

closed 2016-12-04 by marcelstoer