nodemcu/nodemcu-firmware #1640: Populating esp init data not working on `dev`

devsaurus, 2016-12-04:
### Expected behavior
Flash firmware after a full erase works as described in the docs:
```
esptool.py erase_flash
esptool.py -b 460800 write_flash -fs 32m -fm qio 0x00000 bin/0x00000.bin 0x10000 bin/0x10000.bin
```

### Actual behavior
After flashing, the firmware won't boot and is stuck in an endless loop, spitting out stuff at 74480 baud.

I need to explicitly download `esp_init_data_default.bin` to the matching slot in flash to make it work again.

### NodeMCU version
Observed on current `dev`.

Works out-of-the-box on `master`.

I suspect that changes to the section layout in #1566 affected the automatic init_data update from #1527. At least this is the only significant diff between `master` and `dev` atm.

### Hardware
NodeMCU V1.0 with 4 MByte flash.

djphoenix, 2016-12-04:
Yep, I was discovered this tomorrow. Sorry for crashing this feature. The patch is simple: move `init_data` (in `user_main.c`) from `.text` to `.rodata.dram`.

The reason is `irom` now hoovering all `text` that not marked as `iram` (actually all except SDK libs).

devsaurus, 2016-12-04:
Thanks for the hint @djphoenix. PR triggered to resolve the issue according to your guidance.

PR #1643 (2016-12-04): Move init_data from .text to .rodata.dram section
Fixes #1640.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

As per https://github.com/nodemcu/nodemcu-firmware/issues/1640#issuecomment-264708167: Pre-loaded init_data shall be put into `.rodata.dram` section.
closed 2016-12-04 by marcelstoer