nodemcu/nodemcu-firmware #1642: starting http server before

AdrianDroid, 2016-12-04:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
expect the enduser_setup will work as usual.
### Actual behavior
enduser_setup failed to work.
wifi mode stick to 1,
manual setting to 3, will result no page load in browser.

### Test code
```Lua
srv=net.createServer(net.TCP);
srv:listen(80,function () end);
if (srv ~= nil ) then
    srv:close();
    srv=nil;
    collectgarbage();
end
enduser_setup.start();
```
### NodeMCU version
1.5.4.1

### Hardware
node dev kit v3

8<------------------------ END BUG REPORT -------------------------------------

jmattsson, 2016-12-05:
It is not possible to run two TCP servers on the same port/ip combination, by the very design of TCP/IP. This is not a bug. You can run a server on a different port if you want though.

djphoenix, 2016-12-05:
@jmattsson You see that server is closed before smartconfig starts?

jmattsson, 2016-12-05:
\*derp me\* 

No, I read that as being part of the accept callback. /facepalm  Apologies.

Right, then we're probably looking at a case where the socket hasn't been fully released in LWIP before the eus module tries to listen on it again. @djphoenix you've been tinkering quite a bit in this area, do you have any insights?

AdrianDroid, 2016-12-05:
interesting thing is adding timer in between failed too, isn't timing issue?

```lua

print("http stop");
srv=net.createServer(net.TCP);
srv:listen(80,function () 
end);

--2sec later
tmr.alarm(0,2000, tmr.ALARM_SINGLE ,function()
    print("http stop");
    if (srv ~= nil ) then
        srv:close();
        srv=nil;
        collectgarbage();
    end
end);

--5sec later
tmr.alarm(1,5000, tmr.ALARM_SINGLE ,function()
    print("enduser_setup");
    enduser_setup.start();
end);
```

djphoenix, 2016-12-05:
I was reviewed some code in lwip source, and found that listen sockets are closed properly, **but**... sockets that connected **to** servers kept in TIME_WAIT for a long time, and if I create listen sockets before they expires, it fails.

jmattsson, 2016-12-05:
Sockets being kept in TIME_WAIT is expected and per protocol. Something odd seems to be happening with the server sockets though. I'm terribly short on bandwidth right now and not sure when I'd have a chance to dig into this, so I'm going to drag @pjsg into this conversation too.

djphoenix, 2016-12-06:
@jmattsson As I know, TIME_WAIT state is for matching latest TCP stream incoming packets. Should them really break creation of TCP server? I think no, and in another systems like Linux it right.

jmattsson, 2016-12-06:
It should, unless SO_REUSEADDR is specified.

djphoenix, 2016-12-06:
Ah, right. I use this option always by default... So, LWIP have this option (as I see in config), but is it enabled in nodemcu?

djphoenix, 2017-03-06:
I think reason is as noted in #1836, so keep track only for new issue.

closed 2017-03-06 by djphoenix