nodemcu/nodemcu-firmware #1654: Is there a way to determine the SSID of the connected AP (except callback tracking?)

neilyoung, 2016-12-08:
As subject says

Jonathan411, 2016-12-08:
I don't think this is a bug/feature request so this question is probably better asked in StackOverflow.com (but I'm just a noob, so maybe someone not-so-nooby could weigh in on this).

That said,  have you looked at:  wifi.sta.getapinfo()

djphoenix, 2016-12-08:
wifi.sta.getconfig is enough (you maybe also need to check status according to docs)

marcelstoer, 2016-12-08:
See #1010

closed 2016-12-08 by marcelstoer

neilyoung, 2016-12-28:
Thanks. For some reasons I don't get notifications by mail, so sorry for the late reply

marcelstoer, 2016-12-28:
> For some reasons I don't get notifications by mail

Check the "Watch" setting for this repo. Top of any page, slightly right of the center.

neilyoung, 2016-12-28:
Too stupid, was set to "Not watching". Sorry