nodemcu/nodemcu-firmware #1655: Get auto-detected baud rate from uart module

Jonathan411, 2016-12-08:
The uart module auto-detection of baud rate works great, but how you change the parity, stopbits and databits while still using the auto-detection of baud rate feature?

Baud rate auto-detection only works if you don't call uart.setup(), but you have to call uart.setup() to change parity, stopbits or databits.

If you call uart.setup() you have to supply a baud rate, but you don't know what baud rate to supply!  If you supply a baud rate and make the call to uart.setup() it overwrites the auto-detected baud rate with the baud rate supplied.

My ideal solution would be to call uart.setup with only the id parameter, and it would then return the auto-detected baud rate (which could then be used in subsequent calls to uart.setup in which parity, stopbits and databits could be set).

Baud rate auto-detection is an excellent feature, but its value is greatly diminished if we can't find out what baud rate is detected.

marcelstoer, 2016-12-08:
If this is a support request then I'll have to remind you of #1010 but even if it were a feature request it'd apply 😜 

I don't think I understand your problem.

> If you call uart.setup() you have to supply a baud rate, but you don't know what baud rate to supply!

Why don't you know it? It's the same value you configure your upload tool with. OR is this for cases where you communicate with sensor/actor over UART?

Jonathan411, 2016-12-08:
This should be considered a bug (e.g. design flaw).

How do you communicate with sensors (or other devices) over UART when you don't know the baud, parity, databits or stopbits?

The auto-detect baud feature is a good start, but you can't use it to help determine the other settings (auto-detected baud rate is not retrievable).  If we could get the auto-detected baud rate, we could then easily cycle through the other settings until the data is recognizable.

The baud rate has already been detected, we just need a way to get at it.

Jonathan411, 2016-12-08:
I did some digging into the source code for the uart module, and found the following routine in platform.c

```
uart_setup(id);
    return baud;
```

However, when you try the following code:

`print(uart.setup(0))`

You get the error:

> print(uart.setup(0))
> stdin:1: bad argument #2 to 'setup' (number expected, got no value)
> stack traceback:
> 	[C]: in function 'setup'
> 	stdin:1: in main chunk
> 

It looks like the code is there to return the baud rate, but something is not letting us get at it.

pjsg, 2016-12-09:
There is no existing code to get the current baudrate. However, it isn't difficult to add. I have a branch which ought to work -- but I'm not happy with the API -- just using the setup(id) to return the baudrate seems wrong. https://github.com/pjsg/nodemcu-firmware/tree/get-baud   It builds but I haven't tested it.

pjsg, 2016-12-09:
Thinking about it some more, I suspect that uart.setup(0) ought to return all the setup arguments -- i.e. not just the baudrate, but bits, parity etc.

Jonathan411, 2016-12-09:
Thanks very much Philip!

I will do some testing (although it may take awhile as i have never built the firmware from source, i have just used the cloud build service - but it is time i learnt how to do this).

Returning all the current settings would be more useful, but i didn't want to introduce (or request) something totally new;  uart.setup() already returns the baud rate, but it seems redundant that it only returns the baud rate specified in the parameter list.

If you were going to return all the serial port settings, then an api routine like uart.state(id) may be more in keeping with the NodeMCU approach.

Jonathan411, 2016-12-09:
I now have a nodeMCU build environment set up (on a linux box) and used the branch:  [https://github.com/pjsg/nodemcu-firmware/tree/get-baud ](https://github.com/pjsg/nodemcu-firmware/tree/get-baud) to build a new bin file, which i then uploaded to a nodemcu dev kit module.

Here is the output of the test:

```
NodeMCU 0.9.6 build 20150627  powered by Lua 5.1.4
lua: cannot open init.lua
> print(uart.setup(0))
stdin:1: bad argument #2 to 'setup' (number expected, got no value)
stack traceback:
	[C]: in function 'setup'
	stdin:1: in main chunk
> 

```

So something is not quite right.

I have a build and test environment ready now, so i am happy to test anything to help with this issue.

One note:  i use esptool.py to upload the nodemcu firmware and a lua image file to the ESP8266. I have never had any problems with this process, but this time the lua files were incomplete and the wrong file size:

```
----------------------------
cfgForm.htm    : 142848 bytes
configServer.lua : 1919488 bytes
init.lua       : 1082112 bytes
setupConfig.lua : 1100288 bytes
----------------------------
Total file(s)   : 4
Total size      : 4244736 bytes
```
They should have been like this:

```
----------------------------
cfg.lua         : 269 bytes
cfgForm.htm     : 2628 bytes
configServer.lua : 13174 bytes
init.sav        : 4311 bytes
main.lua        : 14192 bytes
newcfg.lua      : 131 bytes
setupConfig.lua : 4298 bytes
setupWifi.lua   : 2233 bytes
sys.log         : 3027 bytes
----------------------------
Total file(s)   : 9
Total size      : 44263 bytes
```

The nodemcu build was clean and so was the esptool upload, so i don;t know why the filesystem was corrupted (don't know if this is indicative of a problem with the branch/build or another issue entirely).

marcelstoer, 2016-12-09:
> NodeMCU 0.9.6 build 20150627  powered by Lua 5.1.4

This doesn't look right. Are you sure you uploaded the right binary?

Jonathan411, 2016-12-09:
The docker build created 2 bin files and 2 map files:

```
root@Stork:~/jsrc/pjsg/nodemcu-firmware/bin# ls -la
total 3788
drwxr-xr-x  2 root root    4096 Dec  9 15:54 .
drwxr-xr-x 13 root root    4096 Dec  9 14:23 ..
-rw-r--r--  1 root root   33152 Dec  9 15:54 0x00000.bin
-rw-r--r--  1 root root  385960 Dec  9 15:54 0x10000.bin
-rw-r--r--  1 root root    4096 Dec  9 14:23 blank.bin
-rw-r--r--  1 root root     128 Dec  9 14:23 esp_init_data_default.bin
-rw-r--r--  1 root root      79 Dec  9 14:23 .gitignore
-rw-r--r--  1 root root  463456 Dec  9 15:53 nodemcu_float_master_20161209-2051.bin
-rw-r--r--  1 root root 1266730 Dec  9 15:53 nodemcu_float_master_20161209-2051.map
-rw-r--r--  1 root root  451496 Dec  9 15:54 nodemcu_integer_master_20161209-2051.bin
-rw-r--r--  1 root root 1240024 Dec  9 15:54 nodemcu_integer_master_20161209-2051.map
root@Stork:~/jsrc/pjsg/nodemcu-firmware/bin#

```
I used the integer bin file in the esptool upload to the ESP8266.

marcelstoer, 2016-12-09:
An unmodified Docker-built binary from the `dev` branch prints this on booting:

> NodeMCU 1.5.4.1 build unspecified powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

The binary from Philip's branch fails to boot for me, I only see gibberish.

Jonathan411, 2016-12-09:
I had to switch the baud rate to 9600 (ESPlorer) and reboot the esp8266 a few times before i got it to work.

marcelstoer, 2016-12-09:
Well, that was kind-of my point. Something is broken in that branch even if the diff doesn't show. I integrated Philip's changes into my clean `dev` branch, rebuilt with Docker and then flashed. The result is more pleasing at 115200:

```
NodeMCU 1.5.4.1 build unspecified powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
lua: cannot open init.lua
> print(uart.setup(0))
115273

```

Jonathan411, 2016-12-09:
That sounds promising!

Can you give me a link to your dev branch and i will do some more testing with it?

Jonathan411, 2016-12-10:
This is most excellent!  I can't wait to try it.  Thanks very much to Philip, and Marcel, for your efforts!

PR #1658 (2016-12-11): Adds uart.check(0) to get the current uart parameters
Fixes #1655.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This adds the `uart.check(0)` to return the current settings of the UART0. It actually reads from the registers (in the driver) to do this, so it will always return the [... cut]
closed 2016-12-11 by marcelstoer