nodemcu/nodemcu-firmware #1656: warnings building luac.cross

eyaleb, 2016-12-09:
### Expected behavior
no errors or warning during the build.
### Actual behavior
Two warnings:
```
$ lua-5.1 tools/cross-lua.lua 
[builder] Forcing rebuild due to configuration change.
[COMPILE] .build/cross-lua/app__lua__lapi.o
[COMPILE] .build/cross-lua/app__lua__lauxlib.o
app/lua/lauxlib.c: In function 'luaL_loadfile':
app/lua/lauxlib.c:635:4: warning: this 'while' clause does not guard... [-Wmisleading-indentation]
    while ((c = c_getc(lf.f)) != EOF && c != LUA_SIGNATURE[0]) ;
    ^~~~~
app/lua/lauxlib.c:636:5: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'while'
     lf.extraline = 0;
     ^~
[COMPILE] .build/cross-lua/app__lua__lbaselib.o
[COMPILE] .build/cross-lua/app__lua__lcode.o
[COMPILE] .build/cross-lua/app__lua__ldblib.o
[COMPILE] .build/cross-lua/app__lua__ldebug.o
[COMPILE] .build/cross-lua/app__lua__ldo.o
[COMPILE] .build/cross-lua/app__lua__ldump.o
[COMPILE] .build/cross-lua/app__lua__lfunc.o
[COMPILE] .build/cross-lua/app__lua__lgc.o
[COMPILE] .build/cross-lua/app__lua__llex.o
[COMPILE] .build/cross-lua/app__lua__lmathlib.o
[COMPILE] .build/cross-lua/app__lua__lmem.o
[COMPILE] .build/cross-lua/app__lua__loadlib.o
[COMPILE] .build/cross-lua/app__lua__lobject.o
[COMPILE] .build/cross-lua/app__lua__lopcodes.o
[COMPILE] .build/cross-lua/app__lua__lparser.o
[COMPILE] .build/cross-lua/app__lua__lrotable.o
[COMPILE] .build/cross-lua/app__lua__lstate.o
[COMPILE] .build/cross-lua/app__lua__lstring.o
[COMPILE] .build/cross-lua/app__lua__lstrlib.o
[COMPILE] .build/cross-lua/app__lua__ltable.o
[COMPILE] .build/cross-lua/app__lua__ltablib.o
app/lua/ltablib.c: In function 'addfield':
app/lua/ltablib.c:137:3: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
   if (!lua_isstring(L, -1))
   ^~
app/lua/ltablib.c:140:5: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'if'
     luaL_addvalue(b);
     ^~~~~~~~~~~~~
[COMPILE] .build/cross-lua/app__lua__ltm.o
[COMPILE] .build/cross-lua/app__lua__lundump.o
[COMPILE] .build/cross-lua/app__lua__lvm.o
[COMPILE] .build/cross-lua/app__lua__lzio.o
[COMPILE] .build/cross-lua/app__lua__luac_cross__luac.o
[COMPILE] .build/cross-lua/app__lua__luac_cross__loslib.o
[COMPILE] .build/cross-lua/app__lua__luac_cross__print.o
[COMPILE] .build/cross-lua/app__lua__..__modules__linit.o
[COMPILE] .build/cross-lua/app__lua__..__libc__c_stdlib.o
[LINK] luac.cross
.build/cross-lua/app__lua__luac_cross__loslib.o: In function `os_tmpname':
loslib.c:(.text+0x27c): warning: the use of `tmpnam' is dangerous, better use `mkstemp'
[builder] Done building target.
```
### Test code
Run the above build `lua-5.1 tools/cross-lua.lua`.
### NodeMCU version
Current dev branch
### Hardware
N/A
### Comment
The first warning looks like the `while` is not indented enough.
The second warning needs investigation to decide if curlies are required or the indentation should be fixed (I think the latter).

TerryE, 2016-12-09:
Both of these are indentation layout issues brought over from the original Lua 5.1.5 base code so we can't blame anyone for introducing them. :smile:

The only issue is whether we tweak the indent layout here to remove the compiler warnings.

eyaleb, 2016-12-09:
@TerryE I expected so.
My attitude is to have no warnings, or else people get used to seeing messages and miss real issues. It really is a good rule for any project.
But this is just me with my R&D manager hat on, and I expect you (and the rest of the developers) have more urgent issues to deal with.

TerryE, 2018-07-02:
Fixed in `dev`

closed 2018-07-02 by TerryE