nodemcu/nodemcu-firmware #1660: EUS does not complete

linuxslate, 2016-12-10:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

EUS Completes, and connects to WiFi.  WiFi AP, and credentials are remembered.

Also, previous EUS web page looked better.

### Actual behavior

Apparently repeated calls to the completion function, followed by a crash.
Unit then scrashes on any subsequent WiFi connection.

### Test code

Code works fine with previous build.

if wifi.sta.status() ~= 5 then  --Bad WiFi

	file.remove("static.lua") -- New Network, delete old settings

	uart.write(0,"\r   CONFIGURE WIFI   ")
	--print("CONFIGURE WIFI")

	wifi.setmode(wifi.STATIONAP)
	wifi.ap.config({ssid="VFD Clock Setup", auth=wifi.OPEN})
	enduser_setup.manual(true)
	enduser_setup.start(
	function()
    --print("Connected to wifi as:" .. wifi.sta.getip())
                  --12345678901234567890
    ssid = string.sub(wifi.sta.getconfig(),1,14)
    uart.write(0,"\rSSID: "..ssid) 
    enduser_setup.stop()
    wifi.setmode(wifi.STATION)
    ssid = nil
	dofile("vfd_clock.lua")
	end,
	function(err, str)
    --print("enduser_setup: Err #" .. err .. ": " .. str)

    uart.write(0,"\r WIFI SETUP ERROR   ")
	end
)

else -- Good WiFi

	enduser_setup.stop()
	wifi.setmode(wifi.STATION)

	--check for static IP settings file.
	if file.exists("static.lua") then
 	  dofile("static.lua")
	end
	uart.write(0,"\r IP "..wifi.sta.getip())
  	dofile("vfd_clock.lua")

end

### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: 7b83bbb2ea134cd85ac9d63108603cc02c4e20f7
	SSL: false
	modules: enduser_setup,file,net,node,pcm,rtctime,sntp,tmr,uart,wifi
 build 	built on: 2016-11-23 14:41
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

### Hardware
"Real" Amica NodeMCU

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2016-12-11:
> branch: master
...
build built on: 2016-11-23 14:41

The EUS module was improved quite a bit with the 1.5.4.1-master_20161201 release. Please try a current build from `master`.

linuxslate, 2016-12-21:
Seems improved using following build:

NodeMCU custom build by frightanic.com
	branch: master
	commit: 81ec3665cb5fe68eb8596612485cc206b65659c9
	SSL: false
	modules: enduser_setup,file,net,node,pcm,rtctime,sntp,tmr,uart,wifi
 build 	built on: 2016-12-21 23:02
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

---
I was able to assimilate to a network, however:

1.  Multiple AP's on the same network show up as multiple networks. It was not this way in older builds. The user has no way of knowing which to select, as signal strength is not shown.

2. Cosmetic/Editorial issues with the default EUS page (Centering of text, etc.)

3. Crash any time    enduser_setup.stop()  is called.  Is it now not allowed to call  enduser_setup.stop()   when it is not running?

4. (Not really EUS, but related to initial and subsequent WiFi Connection):
=wifi.sta.status()
Returns 255 initially -- Before any WiFi Connection. (Before EUS is run).  Is 255 now a defined response?

jfollas, 2017-01-26:
@linuxslate:

1. I'm in the EUS code right now. Not sure what older builds did differently (so maybe it's a SDK 2.0 thing that multiple APs are appearing in the APList?). However, I can modify the default HTML/JavaScript to filter out duplicate APs and show only the strongest (with some sort of signal strength indicator).

2. Do you have a screenshot showing what you think is wrong, or how you think it should look? One thing that is top of mind is that we can't include external libraries or images due to the size of RAM (the entire HTML payload must fit in RAM to be served, unless somebody comes up with a modification to this module to stream content from Flash). 

3. I have a feeling that a lot of the misbehavior that you have been reporting is due to manual mode. Admittedly, I haven't spent much time testing manual mode because all of my scenarios used automatic mode (with the SSID prefix compiled into the firmware). I will be spending a lot more time testing manual mode scenarios on this iteration.

4. I noticed a weird status recently, too, but didn't know what the value was (it was just something that wasn't covered by known values, so the status showed blank on the web page). I have no idea what wifi sta status 255 is, nor how to reproduce it. Are you able to consistently reproduce that status?

linuxslate, 2017-01-26:
Please see #1718.

OK to close this one as a duplicate.

closed 2017-01-26 by marcelstoer