nodemcu/nodemcu-firmware #1663: ESP32: i2c master module

devsaurus, 2016-12-11:
**Rationale:**
I2C master mode is a key feature to attach many external components like sensors or displays.

- [ ] Basic synchronous I2C master support.
  - Lua: The API for basic mode shall maintain back-ward compatibility with existing scripts.
  - Platform: Functions provided to perform init, read & write operations.
- [ ] Asynchronous I2C master support.
  - Lua: API extensions with (optional) callback functions. Lua command finishes immediately and calls back the provided function once transfer terminated.
  - Platform: Expected to establish an interrupt handler with C-callback registration. Synchronous variants for basic support encapsulate IRQ registration and busy wait for the callback.

**Priority:**
  - High for basic support
  - Low for async version

**Dependencies:**
  - I2C driver released with future IDF.

devsaurus, 2017-01-15:
Implementation blocked by espressif/esp-idf#241.

devsaurus, 2017-01-20:
@jmattsson if the issue with the I2C hardware driver isn't fixed anytime soon, I would have to port the software-based master driver as workaround.
Where does a "user driver" go into? E.g. `components/driver` doesn't exist yet.

jmattsson, 2017-01-21:
I'd suggest `components/driver_i2c` in that case. I put the console driver in `components/driver_console` to avoid potential name conflicts with the IDF's `driver` component.

devsaurus, 2017-01-28:
I've ported the sw i2c driver:
https://github.com/devsaurus/nodemcu-firmware/tree/dev-esp32-i2c

Things appear to work on the LA, but I need to check more thoroughly against a real slave.

devsaurus, 2017-02-02:
The basic version is now available in [`dev-esp32` branch](https://github.com/nodemcu/nodemcu-firmware/commit/381726f103b677b81c4b3dfb32e716b1d45aeeda). Feel free to give it a try.

devsaurus, 2017-02-04:
Support for additional 2 hardware channels in synchronous and asynchronous operation is now available as well with https://github.com/nodemcu/nodemcu-firmware/commit/f5632f41f94df2aa1dd72a7cdef9fd454839e64b.
I'm considering this issues as done.

closed 2017-02-04 by devsaurus