nodemcu/nodemcu-firmware #1666: Replace JSON implementation with Streaming version & integrate into HTTP

pjsg, 2016-12-12:
### Missing feature
Incremental (streaming) encoding and decoding of JSON -- especially when sending HTTP posts or parsing HTTP responses.
### Justification
In a memory limited environment, having to read the entire JSON blob and then decode it, and then discard it is not very memory efficient.

### Workarounds
You have to read the response and then decode it.

## Approach

For decoding an HTTP response, use a streaming JSON implementation to read the chunks as they arrive and construct the output object incrementally. If the table object being used is a custom object then it can discard any uneeded parts of the response.

Similarly for encoding, it can be incremental and send the data over the connection as it is generated. With a suitable custom object, the generated string could be larger than memory....

In particular, I'm trying to parse responses from github's api and they are rather large....

marcelstoer, 2016-12-12:
This is also related to https://github.com/nodemcu/nodemcu-firmware/issues/1316#issuecomment-239076741.

jmattsson, 2016-12-13:
Would the SDK JSON support be suitable to build this on? And if so, is it the same that's available in the IDF?

pjsg, 2016-12-13:
The SDK json is cJSON (i.e. what nodemcu uses already). Unfortunately this is not a streaming implementation.

djphoenix, 2016-12-14:
@pjsg please note that text JSON payload is not most-significant memory consumer. The biggest thing is lua-tables, and I absolutely can't imagine how to optimise it.
I was use cJSON to store config on SPIFFS, but it fails with OOM before `encode` invocation (but there was just array of 11-12 hash-tables).

pjsg, 2017-01-19:
There is an interesting issue around how to handle unicode characters. For example, what should

[ "\u1234" ] 

turn into? 

Worse, do we want round tripping to work for arbitrary strings? The issue is whether we consider a string to be 8-bit characters (probably something like ISO-8859-1) or whether it is encoded as utf-8. This makes a difference in both encode and decode. Not all strings of byte are valid utf8 strings, and it may not be possible to encode/decode them and end up with the same thing.

However, if we expect these encoded chunks to be exchanged with other platforms, then it probably makes sense to do the utf8 encoding.

Thoughts?

jmattsson, 2017-01-19:
UTF-8. Definitely. Anything else just leads to more pain in my experience. Well, using a 32-bit `wchar_t` can be less pain in some ways, but oh the memory...

marcoskirsch, 2017-01-19:
By default, JSON is encoded as UTF-8. This is also the only encoding I've encountered in the real world, but my experience with this is limited.
https://en.wikipedia.org/wiki/JSON#Data_portability_issues

pjsg, 2017-01-19:
OK -- I'm convinced. utf8 it is. I need to think about surrogate pairs, but I could just ignore those for the time being.....

pjsg, 2017-01-19:
As far as memory safety is concerned -- the decoder only (currently) does a single calloc call and that is handled appropriately. I'll probably try and fix that to use user data or something similar.

jmattsson, 2017-01-19:
I'm sure Terry will helpfully point you in the direction of `luaM_malloc()` :)

pjsg, 2017-01-19:
I'm doing all the development using a stock x86 version of lua. Nice quick compile/test cycle. The decoder is essentially working. Now on to the encoder (which should be simpler).

pjsg, 2017-01-20:
It now works to both encode and decode (on x86). Need to move it over to nodemcu, and write the documentation....

pjsg, 2017-01-21:
What do you think that the result of encoding {} into JSON should be? There are two choices -- {} or [] . These both decode into the same thing. I favor [], but the existing cjson implementation uses {}. 

I think that you are more likely to want an empty list rather than an empty object.

jmattsson, 2017-01-21:
That's... tricky! While I'd favour backwards compatibility, I also recognise what you say about [] probably being the more likely use-case.

I don't suppose one could pass a configuration option when encoding to say "empty-tables-as-object" or "empty-tables-as-array"?

Oh, and what are you using as `null`? Do you have an `xjson.null` or `xjson.NULL` constant (with `xjson` being whatever you call the module)?

pjsg, 2017-01-21:
I was using nil for null. There are some issue(s) with it (not least being that you don't see nulls at the end of arrays). I'm open to suggestions. Essentially I was doing the first comparisons between cjson and sjson (streaming json) this evening, and I found these differences (so far).

jmattsson, 2017-01-21:
I do believe you need to have an explicit `null` constant, since in JSON there's a clear difference between a member being set to null vs not being there in the first place (compared to Lua which takes a nil value to indicate member removal).

E.g. `{ something: null }` is very different to `{}`, the latter being what would happen if you did `{ something=nil}`; rather a `{ something: sjson.null }` (or `sjson.NULL`) is needed.

pjsg, 2017-01-21:
Indeed. I've now added sjson.NULL (with the same value that cjson uses -- a userdata = 0). However, I was also testing the JSON.lua encoder/decoder (which is written in pure lua). This uses (by default) nil for null, but allows you to specify a string which will be used instead. It suggests "\0" as a good value.

Accordingly (and to make my tests work) I've added an option to allow the user to specify the null value (for both encode and decode). Currently, userdata always gets encoded as null. I can't think of anything better (except maybe throw an exception).

pjsg, 2017-01-21:
In the streaming vein, you can also do things like:

```
print(sjson.encode({function() return {dog="bar"} end}))
```
which returns

```
[{"dog":"bar"}]
```

If there is an attempt to encode a function, then the function is evaluated (with no arguments) and the returned value is encoded in its place. Coupled with the incremental read, and assuming that each file fits in memory (by itself), you could encode the entire contents of the filesystem into a single json. [One function would return a list of functions, each one of which would produce the contents of a particular file. Since these are evaluated one at a time, there is no need to have all the data in memory at once]

pjsg, 2017-01-21:
[Documentation](https://github.com/pjsg/nodemcu-firmware/blob/sjson/docs/en/modules/sjson.md) in my branch. I'm interested in feedback before I submit the PR.

pjsg, 2017-01-21:
The code size is a little larger -- 8k for sjson, and 6k for cjson.

marcelstoer, 2017-01-21:
my $0.02

> What do you think that the result of encoding {} into JSON should be?

I would look at how ECMAScript JSON (i.e. the "source", ECMA-404, RFC 4627) handles that. If you agree that the closest thing in JavaScript to a Lua table is an array then you do

`console.log(JSON.stringify([]))` in the browser console and you'll get `[]`.

> I think that the json NULL should be represented by nil.

Again, you're right. `console.log(JSON.parse("[null, 1]"))` returns `[null, 1]`.

pjsg, 2017-01-22:
What I was trying to get at is that json '[]' and '{}' both end up as a lua empty table. That is fine. The question is what does a lua empty table encode into? It has to be either '{}' or '[]'. I favor the latter.

The null question is because the natural mapping of JSON null is to lua nil. However, lua nil has some special meaning (table values cannot be nil).

fikin, 2017-03-21:
i've read the documentation and sketched a mental exercise how that would play out with streaming io server, something like this one:
```lua
function cb_on_new_net_request(req, resp)
  local jsonDecoder = sjson:decoder()
  req:on("onNextBodyChunk", function(data) jsonDecoder:write(data) end)
  req:on("onDone", function() processRequest( resp, jsonDecoder:result() ) end)
end
```
using current net module this way is possible for custom tcp protocol. for http protocol, a nodejs http-
parser like is needed to wrap the net module first.

to me the decoder part looks ok.
when do you intend of pushing it to master?

marcelstoer, 2017-03-21:
> when do you intend of pushing it to master?

- The release process is documented here: https://github.com/nodemcu/nodemcu-firmware/#releases.
- I already approved the PR #1755 but someone else needs to approve and merge it (going to `dev`).
- It's planned for the next master drop as per https://github.com/nodemcu/nodemcu-firmware/milestone/5.
- However, due to #1810 we might extend the deadline for the next master drop beyond April 1st.

fikin, 2017-03-21:
cool, looking forward. and thanks for the links.

PR #1755 (2017-03-22): Add support for streaming JSON encoder/decoder
Fixes #1666 (partially).

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This makes use of the jsonsl package (which is under a very liberal license). It wraps it in a layer to interface to lua for decoding json strings. There is also a [... cut]
closed 2017-03-26 by marcelstoer