nodemcu/nodemcu-firmware #1667: Provide API meta model

marcelstoer, 2016-12-13:
### Missing feature
A meta model of our API in JSON/YAML/etc format. Contrary to our docs this is not something you want to maintain manually but generate with every build.

### Justification
If want to build code completion aka content assist for NodeMCU into an IDE you need to define the API in some form.

### Workarounds
A user at http://www.esp8266.com/viewtopic.php?f=22&t=12777 resorted to parsing the documentation(!) to build a meta model. Apparently this was a one-off process as he had to manually tweak the model - no surprise there.

hiperiondev, 2017-03-27:
I have converted the zerobrain api to JSON.

It can be used for create a metamodel.

https://github.com/hiperiondev/nodemcu-api-zerobrain/tree/json-documentation

spaccabit, 2018-01-03:
Using [Ldoc](https://github.com/stevedonovan/LDoc/blob/master/doc/doc.md). Ldoc support

- custom formatter markdown,html, ecc..
- custom tags
- Extension modules written in C

pjsg, 2018-01-03:
Can ldoc generate markdown so that we can use it with readthedocs?

spaccabit, 2018-01-03:
In Ldoc you can create custom formatter if the provided formatter is not compatible with readthedocs specifications.

In addition, you could create JSON , XML, etc... formatter for automatic completion of various IDEs.

closed 2019-08-04 by stale[bot]