nodemcu/nodemcu-firmware #1670: time alarms

xhdix, 2016-12-15:
hi 

### Missing feature
time alarms daily weekly ... 

### Justification
 in arduino works good.. and very useful in IOT devices 

### Workarounds
https://www.pjrc.com/teensy/td_libs_TimeAlarms.html

thanks you.

marcelstoer, 2016-12-15:
There's the [timer module](http://nodemcu.readthedocs.io/en/latest/en/modules/tmr/) but due to SDK constraints it maxes out an interval of 1:54:30.947. In light of #1010 I propose to close this.

xhdix, 2016-12-15:
need library similar td_libs_TimeAlarms .. 

in Hardware Requirements, they said :
"TimeAlarms does not require any special hardware, because it uses the time and date provided by the Time library."

but i have to build this 💯  

sorry for my poor(!!) english :D

marcelstoer, 2016-12-15:
> "TimeAlarms does not require any special hardware, because it uses the time and date provided by the Time library."

Sure, except that this library is not available on NodeMCU.

closed 2016-12-15 by marcelstoer

xhdix, 2016-12-15:
classes should be work!! 
run class function check every 1s or 1m with tmr.alarm and then check status!! 
one timer for unlimited timers..

thank you
بدرود