nodemcu/nodemcu-firmware #1672: Better approach to initial contents of filesystem

pjsg, 2016-12-17:
### Missing feature

Initial contents of file system without needing to know the size of of FLASH / generating a huge bin file

### Justification
Makes it more likely that people will flash initial file into the image.

### Workarounds
No

The idea is just to make a C file that contains the contents of the files (probably represented as a ROTable). Then, when the platform auto-formats the filesystem, it copies in the contents of this rotable. It *might* be worthwhile exposing the contents of this table as (say) `initfile`. It might be worth adding a method to copy the current contents back to the filesystem (i.e. a reset to factory).

jmattsson, 2016-12-17:
If you're doing that, I'd rather see those files exposed via a read-only pseudo-filesystem mounted at e.g. /builtin. It'd be trivial to have a copy command in `init.lua` to copy to SPIFFS if that's wanted.

pjsg, 2016-12-18:
I thought about that -- but I really want to be able to preload `init.lua` onto the device and have it bootstrap the rest of the platform. 

I also wondered about a union filesystem where (for example) the spiffs filesystem would overlay the read-only file system. I.e. the read-only filesystem would appear, until a file was written with the same name, and then the new file would appear.

pjsg, 2016-12-18:
I think that the real decision is whether to make this simple or complex. Anything that involves building a new file system (whether or not it is a union filesystem), or changes the way that the `init.lua` is found will probably cause problems. 

We could make the RO File system just be mounted at a new /RO root and have the interpreter search /FLASH and then /RO for files.... But this seems confusing (if you open a file explicitly with file.open, you would have to get the path right...)

devyte, 2016-12-20:
> I really want to be able to preload init.lua onto the device and have it bootstrap the rest of the platform
Isn't this already possible with the spiffs builder? I haven't used this myself, but I thought you could put files in a directory, and the spiffs image gets created, which you then flash. I seem to remember @TerryE mentioning this at some point.

> I also wondered about a union filesystem where (for example) the spiffs filesystem would overlay the read-only file system
I can say from experience that this is a Bad Idea. The main issue is that, at runtime, you never know for sure which file/filesystem is being used, which makes debugging a huge headache. Using a solution like what @jmattsson describes, with a pseudo-builtin fs, would be much better.

My personal opinion is that having files embedded in the firmware itself is not the way to go, because it increases the image size. If you consider OTA updating, where 2 images need to be in the flash, it means that those files are stored x2, plus the copy that is made for modification. 
Instead, it would be better to have a pseudo-directory in the spiffs (i.e.: files with prefix), and have those files "copied" to the "root" for modification as needed. Of course, I'm thinking of a spiffs shared between both images. I've tried this for a "factory-reset"-like behavior, and it works well.

pjsg, 2016-12-21:
I suspect that I have a different use case -- this is really part of my auto-deploy scheme. I have an init.lua that auto deploys from github (given a public repo user/name/branch combination). It then checks regularly for changes to the repo. if there are changes, then it downloads the new code and restarts the nodemcu.

This is not intended for mass deployment, but for cheap and cheerful hobbyists. The only bit of information required to bring up the platform is the repo information. 

I'll think if there is a better way to get the small initial file required into the firmware without having to flash two objects, one of which is very large. It could be that, in this mode, there is another spiffs FS which is directly appended to the end of the firmware and this is copied into the main spiffs on format.
I'd rathr not have to figure out how to resize a spiffs filesystem dynamically (from a small fs to the full size fs on the platform).

closed 2019-06-21 by stale[bot]