nodemcu/nodemcu-firmware #1676: wifi connection can be too  slow

eyaleb, 2016-12-20:
I do not think this is a bug and I am not even asking for an enhancement. I want to understand what is going on and think that people on this forum are most likely to know.
### Expected behaviour
When a wifi connection fails to establish, try again after a **short** delay. Maybe a failure should be returned to the app to let it take action.
### Actual behaviour
There seems to be a constant 1 second delay which is too long for my battery powered gadget.
Is it nodeMCU that implements the 1s delay? Is it the SDK? Can I nominate the delay?
### Test code
Assuming ssid/pass already set and stored.
```Lua
local start_time = tmr.now()
gpio.mode (1, gpio.INPUT, gpio.PULLUP)  -- gpio5 = D1, ground to stop test
if 0 == gpio.read (1) then print ("stopped") else
        local sta = wifi.sta
        sta.setip({ip="192.168.2.59",netmask="255.255.255.0",gateway="192.168.2.7"})
        tmr.alarm(1, 1, 1, function()
                if 5 ~= sta.status() then return end
                tmr.stop(1)
                print (("have connection after %.6fs"):format((tmr.now()-start_time)/1000000))
                tmr.alarm(2, 100, 0, function()         -- 100ms, time for message to print
                        rtctime.dsleep(100*1000)        -- 100ms, whatever
                end)
        end)
end
```
I see about 120ms displayed when all is well, but at times it takes 1.2s or 2.2s.
### NodeMCU version
latest dev
### Hardware
Any module (am using a few).

eyaleb, 2016-12-20:
Me again. I switched to using `sta.eventMon` and got better results.
```lua
-- measure the time it takes for wifi connection to be established using event monitor
local start_time = tmr.now()
gpio.mode (1, gpio.INPUT, gpio.PULLUP)  -- gpio5 = D1, ground to stop test
if 0 == gpio.read (1) then print ("stopped") else
        local sta = wifi.sta
        local function die(reason)
                print (("\n%s: after %.6fs\n"):format(reason, (tmr.now()-start_time)/1000000))
                sta.eventMonStop()
                tmr.alarm(2, 100, 0, function()         -- 100ms, time for message to print
                        rtctime.dsleep(100*1000)        -- 100ms, whatever
                end)
        end
        sta.setip({ip="192.168.2.59",netmask="255.255.255.0",gateway="192.168.2.7"})
        sta.eventMonReg(wifi.STA_GOTIP, function() die("GOTIP") end)
        sta.eventMonReg(wifi.STA_FAIL, function() die("FAIL") end)
        sta.eventMonStart(10)
end
```
I am now running a longer test to see if the timing remains stable.

dnc40085, 2016-12-20:
Hello @eyaleb, I would suggest that you use [`wifi.eventmon.register()`](http://nodemcu.readthedocs.io/en/master/en/modules/wifi/#wifieventmonregister) instead, it uses the SDK's builtin event handler which is much faster than the wifi.sta.eventMonX() functions.

eyaleb, 2016-12-20:
I used an interval of 1ms in my app so the reaction time was very short, but I see your point. My test shows very similar results.
```lua
local start_time = tmr.now()
gpio.mode (1, gpio.INPUT, gpio.PULLUP)  -- gpio5 = D1, ground to stop test
if 0 == gpio.read (1) then print ("stopped") else
        local eventmon = wifi.eventmon
        local function die(reason)
                print (("\n%s: after %.6fs\n"):format(reason, (tmr.now()-start_time)/1000000))
                eventmon.unregister(wifi.eventmon.STA_GOT_IP)
                eventmon.unregister(wifi.eventmon.STA_DISCONNECTED)
                tmr.alarm(2, 100, 0, function()         -- 100ms, time for message to print
                        rtctime.dsleep(100*1000)        -- 100ms, whatever
                end)
        end
        wifi.sta.setip({ip="192.168.2.59",netmask="255.255.255.0",gateway="192.168.2.7"})
        eventmon.register(wifi.eventmon.STA_GOT_IP, function(T) die("GOTIP") end)
        eventmon.register(wifi.eventmon.STA_DISCONNECTED, function(T) die("FAIL") end)
end
```
```
GOTIP: after 0.119389s
GOTIP: after 0.119448s
GOTIP: after 0.119248s
GOTIP: after 0.121174s
GOTIP: after 0.119536s
GOTIP: after 0.119219s
GOTIP: after 0.122006s
GOTIP: after 0.124123s
GOTIP: after 0.130277s
GOTIP: after 0.121471s
GOTIP: after 0.119936s
GOTIP: after 0.124075s
GOTIP: after 0.122032s
GOTIP: after 0.122044s
GOTIP: after 0.119275s
GOTIP: after 0.120650s
GOTIP: after 0.119451s
GOTIP: after 0.119355s
GOTIP: after 0.119455s
GOTIP: after 0.119277s
GOTIP: after 0.125237s
GOTIP: after 0.119284s
```

closed 2016-12-24 by marcelstoer