nodemcu/nodemcu-firmware #1680: MQTT: Connection failure callback not work

alexz-kh, 2016-12-23:
### Expected behavior
Connection failure callback reason should appear

### Actual behavior
Mqtt stuck.

### Test code
```Lua
    m = mqtt.Client("node-dev", 1)
    m:on("offline", function(con)
        print("E:Mqtt down!")
        m:close()
    end)
    m:connect("wf-dongle2.lan", 8883, 0,
      function(client) print("connected") end,
      function(client, reason) print("failed reason: "..reason)
    end)
```
### NodeMCU version
[http://nodemcu-build.com/builds/nodemcu-dev-11-modules-2016-12-23-17-08-53-float.bin](url)
Dev branch, i guess it was commit 1452fbfd1394e4fe9073a2eb22c1a476d258a716
Code with  file, gpio, i2c, mqtt, net, node, pwm, rfswitch, tmr, uart, wifi.

### Hardware
Nodemcu-dev board

### Some debug : 
I found, that mqtt stuck only if try connect to  any *open* port of mqtt-server  - but w\o any listening service on it.
tcpdump result(on server side) proves it :
1) Port open, service down
Btw,  **only one try to connect!**

```
20:58:59.339071 IP (tos 0x0, ttl 255, id 8, offset 0, flags [none], proto TCP (6), length 44)
    nodemcu-dev.lan.36005 > wf-dongle2.lan.8883: Flags [S], cksum 0x3449 (correct), seq 6511, win 5840, options [mss 1460], length 0
        0x0000:  c83a 35d0 7690 18fe 34d6 2f47 0800 4500  .:5.v...4./G..E.
        0x0010:  002c 0008 0000 ff06 377e c0a8 017b c0a8  .,......7~...{..
        0x0020:  017a 8ca5 22b3 0000 196f 0000 0000 6002  .z.."....o....`.
        0x0030:  16d0 3449 0000 0204 05b4                 ..4I......
20:58:59.339395 IP (tos 0x0, ttl 64, id 56923, offset 0, flags [DF], proto TCP (6), length 40)
    wf-dongle2.lan.8883 > nodemcu-dev.lan.36005: Flags [R.], cksum 0x62c2 (correct), seq 0, ack 6512, win 0, length 0
        0x0000:  18fe 34d6 2f47 c83a 35d0 7690 0800 4500  ..4./G.:5.v...E.
        0x0010:  0028 de5b 4000 4006 d82e c0a8 017a c0a8  .(.[@.@......z..
        0x0020:  017b 22b3 8ca5 0000 0000 0000 1970 5014  .{"..........pP.
        0x0030:  0000 62c2 0000                           ..b...
```

2) If i DROP packages from node, with simple rule:
```bash
iptables -A INPUT -i wlan0 -p tcp --destination-port 8883  -j DROP
```
mqtt.client response correctly with, **after 5 connection attempts**:
```
> failed reason: -5
```
Tcpdump result:
```
21:08:41.600489 IP (tos 0x0, ttl 255, id 23, offset 0, flags [none], proto TCP (6), length 44)
    nodemcu-dev.lan.25797 > wf-dongle2.lan.8883: Flags [S], cksum 0x5b71 (correct), seq 6695, win 5840, options [mss 1460], length 0
        0x0000:  c83a 35d0 7690 18fe 34d6 2f47 0800 4500  .:5.v...4./G..E.
        0x0010:  002c 0017 0000 ff06 376f c0a8 017b c0a8  .,......7o...{..
        0x0020:  017a 64c5 22b3 0000 1a27 0000 0000 6002  .zd."....'....`.
        0x0030:  16d0 5b71 0000 0204 05b4                 ..[q......
21:08:42.595240 IP (tos 0x0, ttl 255, id 24, offset 0, flags [none], proto TCP (6), length 44)
    nodemcu-dev.lan.25797 > wf-dongle2.lan.8883: Flags [S], cksum 0x5b71 (correct), seq 6695, win 5840, options [mss 1460], length 0
        0x0000:  c83a 35d0 7690 18fe 34d6 2f47 0800 4500  .:5.v...4./G..E.
        0x0010:  002c 0018 0000 ff06 376e c0a8 017b c0a8  .,......7n...{..
        0x0020:  017a 64c5 22b3 0000 1a27 0000 0000 6002  .zd."....'....`.
        0x0030:  16d0 5b71 0000 0204 05b4                 ..[q......
21:08:43.594872 IP (tos 0x0, ttl 255, id 25, offset 0, flags [none], proto TCP (6), length 44)
    nodemcu-dev.lan.25797 > wf-dongle2.lan.8883: Flags [S], cksum 0x5b71 (correct), seq 6695, win 5840, options [mss 1460], length 0
        0x0000:  c83a 35d0 7690 18fe 34d6 2f47 0800 4500  .:5.v...4./G..E.
        0x0010:  002c 0019 0000 ff06 376d c0a8 017b c0a8  .,......7m...{..
        0x0020:  017a 64c5 22b3 0000 1a27 0000 0000 6002  .zd."....'....`.
        0x0030:  16d0 5b71 0000 0204 05b4                 ..[q......
21:08:44.596151 IP (tos 0x0, ttl 255, id 26, offset 0, flags [none], proto TCP (6), length 44)
    nodemcu-dev.lan.25797 > wf-dongle2.lan.8883: Flags [S], cksum 0x5b71 (correct), seq 6695, win 5840, options [mss 1460], length 0
        0x0000:  c83a 35d0 7690 18fe 34d6 2f47 0800 4500  .:5.v...4./G..E.
        0x0010:  002c 001a 0000 ff06 376c c0a8 017b c0a8  .,......7l...{..
        0x0020:  017a 64c5 22b3 0000 1a27 0000 0000 6002  .zd."....'....`.
        0x0030:  16d0 5b71 0000 0204 05b4                 ..[q......
21:08:45.617655 IP (tos 0x0, ttl 255, id 27, offset 0, flags [none], proto TCP (6), length 44)
    nodemcu-dev.lan.25797 > wf-dongle2.lan.8883: Flags [S], cksum 0x5b71 (correct), seq 6695, win 5840, options [mss 1460], length 0
        0x0000:  c83a 35d0 7690 18fe 34d6 2f47 0800 4500  .:5.v...4./G..E.
        0x0010:  002c 001b 0000 ff06 376b c0a8 017b c0a8  .,......7k...{..
        0x0020:  017a 64c5 22b3 0000 1a27 0000 0000 6002  .zd."....'....`.
        0x0030:  16d0 5b71 0000 0204 05b4                 ..[q......
21:08:46.626752 IP (tos 0x0, ttl 255, id 28, offset 0, flags [none], proto TCP (6), length 44)
    nodemcu-dev.lan.25797 > wf-dongle2.lan.8883: Flags [S], cksum 0x5b71 (correct), seq 6695, win 5840, options [mss 1460], length 0
        0x0000:  c83a 35d0 7690 18fe 34d6 2f47 0800 4500  .:5.v...4./G..E.
        0x0010:  002c 001c 0000 ff06 376a c0a8 017b c0a8  .,......7j...{..
        0x0020:  017a 64c5 22b3 0000 1a27 0000 0000 6002  .zd."....'....`.
        0x0030:  16d0 5b71 0000 0204 05b4                 ..[q......
21:08:47.619849 IP (tos 0x0, ttl 255, id 29, offset 0, flags [none], proto TCP (6), length 44)
    nodemcu-dev.lan.25797 > wf-dongle2.lan.8883: Flags [S], cksum 0x5b71 (correct), seq 6695, win 5840, options [mss 1460], length 0
        0x0000:  c83a 35d0 7690 18fe 34d6 2f47 0800 4500  .:5.v...4./G..E.
        0x0010:  002c 001d 0000 ff06 3769 c0a8 017b c0a8  .,......7i...{..
        0x0020:  017a 64c5 22b3 0000 1a27 0000 0000 6002  .zd."....'....`.
        0x0030:  16d0 5b71 0000 0204 05b4                 ..[q......
```

pjsg, 2016-12-26:
Yes -- this looks like a bug. I suspect that it ought to return the -5 error immediately.

However, what do you think ought to happen if you have the auto-reconnect enabled? When (and how often) should the error callback be called?

karrots, 2016-12-26:
One item I noticed recently which is related to this. If MQTT is called with auto-reconnect you can't actually stop it. If you call disconnect it reconnects. If you disconnect and dereference the object inside the error handler the object goes away but you still get output from the MQTT client object trying to connect then a null object error because the client object doesn't exist anymore.

alexz-kh, 2016-12-26:
@pjsg 
Should be another option, like 'auto-reconnect timeout' ? in worst case, i guess 'keep-alive' option can be reused.

Btw, with auto-reconnect option its continue 10-20 times try to connect:
```
enter socket_connect.
leave socket_connect, heap = 30496.
leave mqtt_socket_reconnected.
enter mqtt_socket_reconnected.
enter socket_connect.
leave socket_connect, heap = 30496.
leavreconnected.
enter mqtt_socket_nected.
enter socket_connect.
leave socket_connect, heap = 30496.
leave mqtt_socket_reconnected.
enter mqtt_socket_reconnected.
enter socket_connect.
leave socket_connect, heap = 30496.
socket_reconnected.
enter sockeenter socket_connect.
```
Then, stdout spaming stops, but in tcpdump i still can see traffic.(i bet, that stdout stuck due debug mode)

@karrots 
what about re-init mqtt ? like
```
m:close()
m=nil
```
Looks like work: (debug output was stop)
```
enter mqtt_socket_reconnected.
enter socket_connect.
leave socket_connect, heap = 42632.
leave mqtt_socket_reconnected.
entem:close()
enter mqtt_socket_close.
leave _disconnected.
leave mqtt_socket_reconnected.
m=nil
enter mqtt_delete.
leave mqtt_delete.
```

pjsg, 2016-12-26:
Yes, the `:close()` should stop the auto-reconnect behavior.

I think that if you have auto-reconnect on, then it should keep trying (until `close()` is called) to connect with no signalling of errors. If the application programmer cares about timeouts, then they can establish their own (with `tmr`) and take appropriate action at that point. This isn't very satisfactory.... 

The alternative is to make the first connection attempt fatal (if it fails) and then all subsequent errors subject to the auto-reconnect setting. I feel that this is maybe a better approach. I.e. if the first connecton attempt fails, then an error is returned (doesn't matter what auto-reconnect is set to). If it gets connected (once), then all subsequent failures will be handled according to auto-reconnect.

I think that I prefer the second approach.

Thoughts?

karrots, 2016-12-27:
It seems reasonable to reconnect if the client gets disconnected after successful connect (this is what I assumed it did). Unsure if it should keep trying if the first attempt isn't successful. Under certain circumstances, this could lead to being locked out. I was creating a handler for the various errors when I ran across the inability to close the connection with auto-connect. 

My intent was to close if say a bad username/password was supplied instead of continually trying.

devyte, 2016-12-27:
A bit of insight, if it helps, and assuming I'm reading this thread right:
If the ESP is on the same power circuit as the mqtt broker, then both will
power up at the same time after a power failure. The ESP, being a simple
device, will most likely come online a lot sooner than the broker, will try
to connect, and fail.

In such a scenario, it makes sense to keep trying until successful, even if
the first attempt fails.

Given that at least one such scenario exists, it can be argued that it
makes sense to simplify all error handling and just keep trying no matter
which error it is that is causing the fail.

As for what should happen after several failed attempts, I'd say that's up
to the user: send an email, a notification, sms, keep on trying forever, or
just close and pack it up.

karrots, 2016-12-27:
Just about any scenario works if two things get worked out:

1. The ability to stop auto-connect without destroying the object.
1. Ability to destroy the object and have auto-connect actually stop.

I was only suggesting it would only be wise to stop auto-connect under certain error conditions like bad username and password. There are about 10 reasons returned in the error handler. Looking quickly 3-5 of them don't make any sense to retry automatically at least not without exponential backoff.

joysfera, 2017-01-03:
I have just run into the reconnect with wrong name/password issue. It tried to reconnect several times per second like crazy, basically flooding my internet line and the broker. I don't like it. The reconnect should be way slower, the delays between individual attempts should be tens of seconds at least, IMHO. Reusing the keep-alive configurable value seems like the best fix for now.

alexz-kh, 2017-03-14:
Folks, any updates? 
Should i close those one and redirect to #1772 ?

PR #1683 (2017-04-04): Fix the error callback from not being called sometimes
Fixes #1680 

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

There were a number of issues (now fixed);

* If the socket failed to connect before the timer went off, then it wasn't reported. It now is
* If the timer went off bef [... cut]
closed 2017-04-04 by marcelstoer