nodemcu/nodemcu-firmware #1681: u8g Unable run setFont

ycbwb, 2016-12-24:
### Expected behavior
        oled can show something
### Actual behavior
Unable run setFont

lua: init_oled.lua:10: bad argument #1 to 'setFont' (font data expected)
stack traceback:
	[C]: in function 'setFont'
	init_oled.lua:10: in function 'Init_oled'
	init.lua:2: in main chunk
	[C]: ?

### Test code
```Lua
function Init_oled()
   spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, spi.DATABITS_8, 0)
   disp = u8g.ssd1306_128x64_hw_spi(cs, dc, res)
   disp:setFont(u8g.font_6x10)
   disp:setFontRefHeightExtendedText()
   disp:setDefaultForegroundColor()
   disp:setFontPosTop()
end
```
### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	modules: cjson,dht,file,gpio,http,i2c,mqtt,net,node,ow,pwm,spi,tmr,u8g,uart,wifi,file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2016-12-24 12:48
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

### Hardware
Nodemcu-dev board
0.96 oled

pjsg, 2016-12-24:
Having had the same issue myself, the problem is (nearly always) that the parameter passed to setfont is null. 

Check that you have built the correct fonts into the firmware.

Then check that `u8g.font_6x10` is actually non-null. Remember that Lua will return null for any unknown table entry.

ycbwb, 2016-12-24:
@pjsg Thank you  
Has been re apply firmware

In addition to the default u8g.font_6x10    I added u8g.font_5x8

Waiting for firmware build ok...

I am a novice. I don't know how to check the correct fonts.

Excuse my poor  English...

devsaurus, 2016-12-27:
@ycbwb did the new build work for you or are there still issues?

ycbwb, 2016-12-27:
@devsaurus 
Yes it's ok 
Thx

closed 2016-12-27 by ycbwb