nodemcu/nodemcu-firmware #1686: pinf -f crashes nodeMCU imideatelly

hisnameishb, 2016-12-28:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
not rebooting

### Actual behavior
rebooting in under one second.

### Test code
```shell
ping -f ipaddress of nodeMCU
```
### NodeMCU version
1.5.4.1

### Hardware
nodeMCU

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
configurable behaviour on receiving pings:
ping back or ignore pings

### Justification
because of the above behavior

### Workarounds
none

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2016-12-29:
I can't confirm that. I tested on macOS against an old 1.5.4.1 from April and the latest `dev` and both withstand the ping flood.

```
~/Data/NodeMCU/nodemcu-firmware (dev) > sudo ping -f 192.168.0.46
PING 192.168.0.46 (192.168.0.46): 56 data bytes
..Request timeout for icmp_seq 1562
..Request timeout for icmp_seq 8811
..Request timeout for icmp_seq 19651
.^C
--- 192.168.0.46 ping statistics ---
29514 packets transmitted, 29510 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 1.297/2.620/14.798/1.135 ms
```

hisnameishb, 2016-12-30:
It was a power problem. With good power supply theres no rebooting or packet loss.

closed 2016-12-30 by hisnameishb