nodemcu/nodemcu-firmware #1687: i2c slave mode?

mahesh2000, 2016-12-29:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Does the I2C library of nodemcu (lua) support slave mode? i'm not able to get it to work. i want to send data to it from a nordic semi nRF24LE1 which needs to be in master mode. it's (a low-cost, low-power, 8051-based SoC with a long range 2.4GHz proprietary digital radio protocol). Thanks.
### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2016-12-30:
This ain't a bug (report), see #1010.

closed 2016-12-30 by marcelstoer

wolfgangr, 2017-02-08:
Well, it should have been filed as a feature request, of course.
I considered to do so, but since at hte moment I can't contribute, it only could serve as a condensation core.
For this purpose the post at hand with the perfect headline will do as well.

Results from a google search evalutation:

----------------------------

["I have done it"](https://github.com/esp8266/Arduino/issues/1330#issuecomment-276210394) This is not nodemcu, but still ESP8266 SDK, so if true, this might be good point to start.
Further ressources of the same solution:
-> http://www.esp8266.com/viewtopic.php?f=33&t=13546&start=4
-> https://github.com/bjoham/esp8266_Arduino/tree/i2c_slave

----------------------------------------

"rudi" at http://bbs.espressif.com/viewtopic.php?t=2092#p6643 claims:
> .... used the master code and changed simply from clocking to listen the master generated clock
an ISR on SDA gives me the START signal for checking HIGH SCL
hope this helps for other who are searched the same.
**solved**!
here are working just in time 2 esp8266 and one esp32beta ,
one 8266 as master the seconds as slaves.
they are work at standard 50kHz and 120 kHz stable, 250 and 400 kHz sometimes spikes, .....

------------------------------------

http://arduino.stackexchange.com/questions/14487/esp8266-as-i2c-client

> The issue with the ESP8266 is that it is multitasking, and only 80% of the CPU time is available to developers. This manifests as an **outage of 100-200us every 500ms** or so, which rather complicates all sorts of software serial interfaces, such as I2C, SPI and UART. As a bus master the ESP8266 hides these issues as it is responsible for the bus clock, but **as a bus slave it needs to halt data transfer** for the duration of these random events.

> >    It seems that the Expressif SDK only supports master mode.

> Sorry no. Espressif SDK supports Slave Mode too. .... rudi ....

---------------------------------

So my impression is: it should be possible, but there may be quirks.
I remember (can't find the source any more)  that the master has to implement "clock stretching" to accommodate the inevitable time lags. 

From another project, I remember that raspberry suffers from some bug in clock stretching.
So for my plan (NodeMCU as slave for RasPI) I'll postpone "I2C slave".