nodemcu/nodemcu-firmware #1691: node.compile() fails

vsky279, 2016-12-31:
### Expected behavior
`node.compile()` should not fail when compiling correct lua script.

### Actual behavior
`node.compile()` fails to compile correct lua script giving this error
```
> node.compile("watertank.lua")
stdin:1: value too big or small for target integer type
stack traceback:
	[C]: in function 'compile'
	stdin:1: in main chunk
```
### Test code
The file [`watertank.lua`](https://github.com/vsky279/nodemcu-firmware/blob/watertank/lua_examples/watertank/watertank.lua) does not compile. Non-compiled version runs correctly.

### NodeMCU version
Current dev branch.
Included modules: adc, cjson, enduser_setup, file, gpio, http, net, node, ow, rtcmem, rtctime, sntp, tmr, uart, wifi

### Hardware
ESP8266-ESP01, ESP8266-ESP12E

FrankX0, 2017-01-01:
The issue is caused by the constant in:
`if delay < 0 then delay = delay + 2147483647 end`

The interpreter handles it correctly (as already observed by you), but the compiler fails.
2147483647 in hex: 0x7FFFFFFF
Using 2147483646 (0x7FFFFFFE) compiles fine.

This seems a compiler bug to me.

jmattsson, 2017-01-02:
@pjsg or @TerryE, this sounds like your area of expertise

TerryE, 2017-01-02:
@jmattsson I'll take a quick look tonight.  Still wading through Lua 5.3.3, but getting their slowly.

TerryE, 2017-01-02:
@vsky279 what build version are you using and is it an FP or integer build.  IIRC, we had this one a couple of years ago.  I've just tried it with a recent master build and it works fine.

vsky279, 2017-01-02:
@Terrye I'm using the integer build of current dev. I experienced it also on some older version too but I have not reported it at that time.

TerryE, 2017-01-03:
OK.  I'll have to build `luac.cross` on an integer build.  A job for tonight.

pjsg, 2017-01-23:
This is worse than I thought:

```
function f(x) 
  return x + 2147483647
end
```
I put this in a the file `add.lua` and then ran `node.compile('add.lua')` -- and the watchdog rebooted the platform.

As expected, the +2147483646 version compiled just fine. 

The version with + 3147483647  also reboots the platform. However, when executed directly, it is clearly adding 2147483647 to the input number.

[Integer build on dev]

EgorOA, 2017-02-27:
Is this issue fixed or not?

TerryE, 2017-02-27:
No.  Sorry.  I'll try to bump it up the priority list, but this is really something that falls in the "can easily workaround" category.

oyooyo, 2017-11-11:
I just came across a problem that seems to be closely related to this issue.
What's confusing me is that in my case, node.compile() seems to work fine, but luac.cross doesn't.

Here's an example file called test.lua:
```lua
duration = function(start, stop)
  local duration = stop - start
  if duration < 0 then
    duration = duration + 0x80000000
  end
  return duration
end

print(duration(0x7FFFFFFF, 0))
```
If I upload this file as test.lua and execute `require("test")` then the correct result `1` gets printed.
If I compile the file with `node.compile("test.lua")`, remove the original source code file with `file.remove("test.lua")`, reset the device and run `require("test")` again, it still prints the correct result `1`.
But if I compile "test.lua" on my PC, with the luac.cross compiler obtained by running `lua tools/cross-lua.lua` as stated in the documentation, using command `luac.cross -s -o test.lc test.lua`, upload the compiled `test.lc` and once again execute `require("test")`, the wrong value -4294967295 gets printed.

I assume luac.cross actually works fine as well, but that I must pass the correct `-cci` or `-ccn` command line arguments to luac.cross? What's the correct value for these arguments?

closed 2019-08-04 by stale[bot]

vsky279, 2020-08-25:
The issue seems to be fixed. This has been tested on current `dev` branch under Lua 5.1 as well as Lua 5.3.
I think the issue can closed now.

jmattsson, 2020-08-26:
Nice! Thanks for testing, @vsky279 !