nodemcu/nodemcu-firmware #1693: Uart.On data event encode/decode problem

young310, 2017-01-01:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
The same data print by CH340G usb2serial  1B C3 C4 DB 00 1B 1B FF FF 

### Actual behavior

But nodemcu out 4 unreadable something like diamonds
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
uart.on("data", 8,
  function(data)
    print("receive from uart:", data)
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
branch: master
commit: 7b83bbb2ea134cd85ac9d63108603cc02c4e20f7
	
### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

devsaurus, 2017-01-01:
What is the output of
```lua
function (data)
  print("receive from uart:", data)
  print(data:byte(1))
  print(data:byte(2))
  ...
  print(data:byte(8))
end
```
Does it match the input from the terminal?

young310, 2017-01-03:
Thanks! It works now, prints the dec value.
But how can I use the feature: end_char? 
ex: uart.on("data", "??", function(data)

devsaurus, 2017-01-03:
Refer to the docs http://nodemcu.readthedocs.io/en/dev/en/modules/uart/#uarton for flavours of `uart.on()`.
Furthermore, please consider #1010 for support questions.

closed 2017-01-03 by devsaurus

young310, 2017-01-04:
Hi, because the data can be delivered in diverse data types, how the comparison will work if I pass digit/byte in?