nodemcu/nodemcu-firmware #1699: Secure MQTT does not try connecting at all

joysfera, 2017-01-02:
### Expected behavior
MQTT Client connect("broker", 8883, 1) should connect to the broker

### Actual behavior
It doesn't even try to initiate the connection and just fails with reason = -5 (mqtt.CONN_FAIL_SERVER_NOT_FOUND)

### Test code
```Lua
--  mqtt_client:connect("broker", 1883, 0, function(client)
    mqtt_client:connect("broker", 8883, 1, function(client)
        print("MQTT connected")
    end, function(client, reason)
        print("MQTT connect failed: " .. reason)
    end)
```

By uncommenting the line with port 1883 and commenting out the other secure connection line I can get immediate connection to broker so DNS, path to broker and broker itself work OK.

### Broker
I use Mosquitto carefully configured to use tlsv1.1 to please NodeMCU that is limited to TLS v1.1 only (master branch, at least). The certificates are from Let's Encrypt but I didn't load them to NodeMCU. Instead, I relied on the ```net.cert.verify(false)```

### TCPDump
I ran tcpdump at the broker to see if NodeMCU tries to connect at all. I can confirm that there was no attempt - zero bytes transferred, the connection wasn't even initiated. Another test client (an Android app named MQTT Dash) connected to broker just fine so the port is open, broker is listening there and no firewall issues are in the way.

### NodeMCU version
Tried both current master and dev (61d20a75e24ff0bd5c90acf2b207847c7cea9970), same behavior.

### Hardware
Wemos Mini

geedubess, 2017-03-18:
It took a while to find your post, but I'm relieved to hear I'm not the only one with this problem.

I just tried loading a fresh build from branch dev (nodemcu-build nodemcu-dev-13-modules-2017-03-18-07-18-01-integer.bin) with mqtt/tls enabled. Saw normal dns + MQTT packets transmitted with [secure=0], saw 0 packets transmitted with [secure=1] and my callback.

Funky: if I invoke the mqtt connect without callback, I get further - I even see packets transmitted!
I do not recognize the error logged, though - sounds like a legit bug:

> m = mqtt.Client (MQTT_CLIENTID, 120, MQTT_BRUSER, MQTT_BRPWD)
> m:connect ("mqttServer", 8883, 1)
> E:M 1584
E:M 1584
E:M 1584
E:M 1584
E:M 1584
E:M 1584
E:M 1584

> 
--
08:55:25.289128 IP esp8266.10011 > router.domain: 44816+ A? mqttServer. (26)
08:55:25.289723 IP router.domain > esp8266.10011: 44816* 1/0/0 A 192.168.1.102 (42)
08:55:25.291890 IP esp8266.3125 > mqttServer.8883: Flags [S], seq 6512, win 5840, options [mss 1460], length 0
08:55:25.293615 IP mqttServer.8883 > esp8266.3125: Flags [S.], seq 1467678902, ack 6513, win 14600, options [mss 1460], length 0
08:55:25.301341 IP esp8266.3125 > mqttServer.8883: Flags [F.], seq 1, ack 1, win 5840, length 0
08:55:25.305789 IP mqttServer.8883 > esp8266.3125: Flags [F.], seq 1, ack 2, win 14600, length 0
08:55:25.307117 IP esp8266.3125 > mqttServer.8883: Flags [.], ack 2, win 5839, length 0
08:55:30.293348 ARP, Request who-has esp8266 tell router, length 28
08:55:30.350862 ARP, Reply esp8266 is-at a0:20:a6:02:77:41 (oui Unknown), length 28

joysfera, 2017-03-18:
I don't know anything about NodeMCU internals but I guess the "E:M" is an error saying that memory is low. That would explain why you get further without a callback - as the callback consumes additional memory.
I recall that I managed to send a packet by using IP address instead of a domain name in the MQTT - again the IP address avoided DNS thus saved a few bytes of precious memory. Still not enough for a reliable work.
I gave up on a secure MQTT (or even HTTP) long time ago. The only hope is ESP32 (due to its much larger RAM) but there's no NodeMCU for ESP32 so we're pretty much ... stuck.

marcelstoer, 2017-03-18:
Is this something that is caused by the 2.0 SDK i.e. how's the behavior on the frozen 1.5.4.1 branch?

joysfera, 2017-03-18:
Given that my original issue is dated Jan 3 while NodeMCU with SDK 2.x was released a month later this issue exists in SDK 1.5.4.1 branch as well.

NicolSpies, 2017-03-19:
I  can confirm issue has always been present resulting in using MQTT not secure since I started.

marcelstoer, 2017-03-19:
Looks like I'm the lucky one here because it does (still) work for me, the client connects just fine. I have a free account with cloudmqtt.com (hosted on AWS). Their certificate is a wildcard cert for *.cloudmqtt.com, details available at https://www.ssllabs.com/ssltest/analyze.html?d=www.cloudmqtt.com&s=52.84.3.134&latest (TLS 1 - 1.2).

Build info:
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 5feae3fee1debe6aef75b8836fa86f9967037f2f
	SSL: true
	modules: cron,dht,file,gpio,http,mqtt,net,node,rtctime,sntp,spi,tmr,uart,wifi,tls
 build 	built on: 2017-02-26 20:37
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
```

geedubess, 2017-03-20:
@marcelstoer or others - if you happen to have a small lua example that you can verify works (including embedded public or disposable cert?), I'd love to try it! I wonder if there's something in my cert that's causing it to blow up, as I'm blindly generating certs without much prior knowledge.

NicolSpies, 2017-03-23:
@marcelstoer, you are not the only lucky one, MQTT secure connection works with the new TLS module. I downloaded the cloudmqtt wildcard certificate and uploaded into ESP as per NodeMCU documentation.

geedubess, 2017-03-24:
I've tried using the cloudmqtt cert - same result. Again, for those of you that have had success, @NicolSpies - would you mind sharing a functional example? cloudmqtt would be perfect!

NicolSpies, 2017-03-24:
@geedubess let's take it off the list, email me direct

NicolSpies, 2017-03-24:
@marcelstoer, @djphoenix.  Interesting observation related to heap usage between secure and non-secure MQTT connection using the same code:
-  **Secure MQTT** connection utilizing TLS module **uses 17.4 kB**
-  **Non-secure MQTT** connection not utilizing TLS module **uses 1.2 kB**

The heavy user could be the TLS module. Another approach could be to use a non-secure connection, if the broker allows it, and encrypt the payload. This might save on heap usage but still leave a security risk ?

NicolSpies, 2017-03-24:
@joysfera Could this issue be closed ?

joysfera, 2017-03-24:
Give me few hours to test it again, against different servers.

joysfera, 2017-04-04:
I have just tried it again, against a public server to avoid possible issue on my broker setup:

```
NodeMCU custom build by frightanic.com
	branch: master
	commit: 22e1adc4b06c931797539b986c85e229e5942a5f
	SSL: true
	modules: file,gpio,mqtt,net,node,tmr,uart,wifi,tls
 build 	built on: 2017-04-04 22:11
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
lua: cannot open init.lua
> m=mqtt.Client(node.chipid(), 120)
> m:connect("iot.eclipse.org", 8883, 1, 0, function(client) print("connected") end, function(client, reason) print("failed:"..reason) end)
> connected
```
So it's working, great! 
Unfortunately, I don't have the extra 16.2 kB on heap for TLS in my app (as @NicolSpies points out above). That might be the reason why secure MQTT was failing for me originally.

I think this issue can be closed now. I'm waiting for ESP32 that should resolve these issues by its 512 kB of RAM.

joysfera, 2017-04-04:
BTW, this is the log on my mosquitto setup when trying to connect from NodeMCU:
```
1491346571: New connection from 1.2.3.4 on port 8883.
1491346572: OpenSSL Error: error:1408F119:SSL routines:ssl3_get_record:decryption failed or bad record mac
1491346572: Socket error on client <unknown>, disconnecting.
```
NodeMCU sort of hangs and prints neither "connected" nor "failed". Seems like I need to finetune the TLS somehow.

joysfera, 2017-04-04:
Final note about my mosquitto setup: I had enforced tlsv1.1 originally because the SDK 1.5.x firmware supported tlsv1.1 only, IIRC. The new mbed TLS supports other versions as well so when I noticed it at last I removed the tlsv1.1 instruction from mosquitto config and voila - NodeMCU now connects to my broker properly. Now just tell me where to get the required +16 kB of heap... :)

djphoenix, 2017-04-05:
I'll go into #1700 soon. Maybe it will reduce required memory for TLS connection. Anyway, how much heap you have *after* all parts (except mqtt) started up?

joysfera, 2017-04-05:
It oscillates between 10 and 20 kB of free heap. The oscillations are caused by slow releasing of web server TCP connections (there's an open issue for that as well) that looks like a memory leak. It eventually recovers a bit for a while (until new connections arrive). I recently split my application into many small modules that are loaded on-the-fly with dofile() to save as much memory as possible. I see no other possibility for me to release more memory on heap.

djphoenix, 2017-04-05:
> slow releasing of web server TCP connections (there's an open issue for that as well)

Nope, it was (almost) fixed in #1838.
> I recently split my application into many small modules that are loaded on-the-fly with dofile() to save as much memory as possible

It's best practice for NodeMCU, I have reached this solution for me. Only callbacks kept in-memory, and callbacks only performs `dofile` for sources that contains logic.

About webserver - are you using Lua implementation or some C-level module? There is #1850 with first try to get native implementation...

joysfera, 2017-04-05:
The web server is my own simple and not very complete Lua implementation. I also have an own DS18B20 Lua implementation, then it's using the JSON module that is known to be memory hungry and many more things. Maybe the new streaming JSON will help me as well.
And the #1838 is in dev only while I use master for production.
I apparently haven't managed to empty the memory enough to have there callbacks only. Large bits of programming logic is still there, plus data structures and what not.

djphoenix, 2017-04-05:
...also. About your "production" version - are you use precompiled versions of sources (.lc) or original (.lua)? And so have you used `node.stripdebug` for compaction of sources? It reduces some RAM (used for store debug symbols)...

closed 2017-04-05 by marcelstoer

joysfera, 2017-04-05:
@djphoenix since all my sources are compiled (.lc) I don't bother removing the debug symbols as the node.compile() is supposed to remove them, anyway.
I'm wondering if there's some tips&tricks documentation for using the dofile() (as official FAQ mentions require("") only). For example, how to call dofile("main.lc") from init.lua so that the init.lua does not stay in memory? I no longer need the init code but I am afraid that it stays there forever.

jrheling, 2018-03-31:
I had a similar experience using 1.5.4.1-final to connect via MQTT w/ TLS to mosquitto 1.4.12.

While I had "tls_version tlsv1.2" included in my mosquitto.conf, any attempt to connect would immediately create an error on the server side like:

> Client connection from 172.16.78.54 failed: error:1402710B:SSL routines:ACCEPT_SR_CLNT_HELLO_C:wrong version number.

This is with the mbed TLS library, wo I'd (perhaps incorrectly) thought TLS 1.2 was supported. In any case, once I commented out the tls_version line, everything seems to work.

marcelstoer, 2018-04-01:
1.5.4.1-final does not use mbedTLS but axTLS which is limited to TLS 1.1: https://nodemcu.readthedocs.io/en/1.5.4.1-final/en/modules/net/

joysfera, 2018-04-01:
One year ago I suggested to close this issue because I got it working (on Apr 5) with a completely empty NodeMCU to avoid low memory issue.
I have just tried it again, just to be sure, with the very latest dev, completely empty, no extra modules:
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 9af4f58417103a21d31f70fc6e44bfba55d36789
	SSL: true
	modules: file,gpio,mqtt,net,node,tmr,uart,wifi,tls
 build 	built on: 2018-04-01 07:22
 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
lua: cannot open init.lua

> m=mqtt.Client(node.chipid(), 120)
> =node.heap()
41232

> m:connect("mybroker", 8883, 1, 0, function(client) print("connected") end, function(client, reason) print("failed:"..reason) end)
> connected

> =node.heap()
24328
```

So it still works fine, still requires 17+ kB of memory on the heap so I think I'll start using it once @TerryE 's LFS frees enough memory.
And again, this issue can be closed, I believe.

marcelstoer, 2018-04-01:
> And again, this issue can be closed, I believe.

It _has_ been closed for nearly a year 😉

joysfera, 2018-04-01:
@marcelstoer see above, sorry again.

joysfera, 2018-04-01:
BTW, I've got an idea how to save memory with TLS connections long time ago that I haven't tested yet (still on my TODO list): using ECC SSL certificate on the server could save quite some memory on client, perhaps? https://wiki.openssl.org/index.php/Elliptic_Curve_Cryptography

jrheling, 2018-04-01:
@joysfera Sorry for the confusion - I commented in order to give future folks troubleshooting this a better chance of finding this issue.

While I had this working on 1.5.4.1-final, after moving to 2.1.0 (116b762) I'm unable to get cert validation to work. Out of curiosity, does your recent example (above) work if you do tls.cert.verify(true) before connecting to the broker?

TerryE, 2018-04-01:
I have been thinking that since HTTPS /TLS on ESP8266 is _so_ marginal in memroy terms that it sort of makes using LFS compulsory, and if this is really the case then being able to use LFS features properly will really simplify the SSL implementation.