nodemcu/nodemcu-firmware #1701: Write better documentation for UDP handling

marcelstoer, 2017-01-03:
Explain the main ideas in the ["net.udpsocket Module" chapter](http://nodemcu.readthedocs.io/en/dev/en/modules/net/#netudpsocket-module) right before the documentation of the individual functions.
Furthermore, the sentence "received callback have port and ip after data argument." isn't correctly formatted (bold?).

marcelstoer, 2017-01-26:
@djphoenix I added a few notes, pls see http://nodemcu.readthedocs.io/en/latest/en/modules/net/#netudpsocket-module

I would appreciate if you could add a standalone example, maybe below the `net.udpsocket:send()` documentation? That one is _really_ terse.

marcelstoer, 2017-02-01:
Furthermore, `net.udpsocket:send()` is lacking important parameter documentation. I figured it out through some trial and error.

I kindly ask someone who knows that stuff to now also review 0ca4b6b and then close this issue if all is well.

This and #1771 are the last pending issues for the next release.

djphoenix, 2017-02-01:
@marcelstoer good work!

closed 2017-02-01 by djphoenix

eyaleb, 2017-02-02:
re https://github.com/nodemcu/nodemcu-firmware/issues/343.

Now that we use SDK 2.0.0 in dev, is there a reliable way to send a UDP packet and then dsleep **after** it is sent (send queue is empty, naturally with UDP there is no concept of the packet delivered)?

I am still unhappy about the hack I use, which may not work if a longer delay is needed due to other SDK (or other) changes in the future.

djphoenix, 2017-02-02:
There is no callback for *actual* UDP send (that means that packet was passed to radiomodule and this have passed it on air).

eyaleb, 2017-02-02:
@djphoenix and I guess no way to check the status of the radio send buffer (is empty)?

djphoenix, 2017-02-02:
@eyaleb unfortunately, no way.

pjsg, 2017-02-02:
Also, note that since 802.11 is an acknowledged protocol, there can be a significant time delay before the frame gets acknowledged. 

I took a quick look, and I couldn't find an upper bound for the time that the MAC might take to send a data frame.

commit f9c9eff7 (2017-01-26): Add a few notes about UDP handling, fixes #1701