nodemcu/nodemcu-firmware #1703: Building ESP32 README

luismfonseca, 2017-01-05:
I've tried to build the `dev-esp32` branch using `make all`.

It builds the binary and I can flash it to the ESP32. However, it got built without any module, other than the built-in Lua modules (i.e. `math`).

I'm using the Mac OS SDK binaries so I'm not sure if that has something to do with it. It looks like the KConfig definitions are simply ignored?

At any rate, it would be nice to have a small README on how to build the `dev-esp32` branch.

marcelstoer, 2017-01-05:
Does it help to look at how the CI build is set up?
-> https://github.com/nodemcu/nodemcu-firmware/blob/dev-esp32/.travis.yml

LiweiDong, 2017-01-16:
When the  bluetooth modules for ESP32 can be released ?
[https://github.com/espressif/esp-idf/blob/master/components/bt/bt.c](https://github.com/espressif/esp-idf/blob/master/components/bt/bt.c)

bwhitman, 2017-01-18:
Having similar issues to @luismfonseca. On OS X, I am doing make menuconfig to ensure the modules are checked, but when flashing I get a working lua prompt but no modules. For example, 

`
Mounting flash filesystem...
mount res: -10025, -10025
Formatting file system. Please wait...
mount res: -10025, -10025
Formatting: size 0x2f0000, addr 0x110000
mount res: 0, 0
tcpip_task_hdlxxx : 3ffbb3bc, prio:18,stack:2048
Task task_lua started.
NodeMCU ESP32 build 20170118 powered by Lua 5.1.4
lua: cannot open init.lua
> Heap size::244604.
print(wifi.sta.getip())
stdin:0: attempt to index global 'wifi' (a nil value)
> 
`

I've tried compiling using the travis command lines pointed out by @marcelstoer : 
`
export BUILD_DATE=$(date +%Y%m%d)
env BUILD_DIR_BASE=build/integer make MORE_CFLAGS="-DLUA_NUMBER_INTEGRAL -DBUILD_DATE='\"'$BUILD_DATE'\"'" defconfig all
 ../esp-idf/components/esptool_py/esptool/esptool.py --port /dev/tty.usbserial-DN02B17H erase_flash
python /Users/bwhitman/outside/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/tty.usbserial-DN02B17H --baud 921600 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /Users/bwhitman/outside/esp/nodemcu-firmware/build/integer/bootloader/bootloader.bin 0x10000 build/integer/NodeMCU.bin 0x8000 build/integer/partitions_singleapp.bin
`

Which again runs the lua prompt fine but same issue. Board is a sparkfun ESP32 "thing."

bwhitman, 2017-01-18:
Interestingly, if I build it on my linux VM I get a bit farther:

```
Mounting flash filesystem...
Formatting file system. Please wait...
tcpip_task_hdlxxx : 3ffbc6bc, prio:18,stack:2048
I (78428) wifi: frc2_timer_task_hdl:3ffbdf6c, prio:22, stack:2048
I (78518) wifi: Init lldesc rx mblock:25
I (78518) wifi: Init lldesc rx ampdu len mblock:7
I (78518) wifi: Init lldesc rx ampdu entry mblock:4
I (78518) wifi: pp_task_hdl : 3ffcae58, prio:23, stack:8192
NodeMCU ESP32 build 20170118 powered by Lua 5.1.4
lua: cannot open init.lua
> print(wifi.sta.getip())
stdin:1: attempt to call field 'getip' (a nil value)
> wifi.setmode(wifi.STATION)
stdin:1: attempt to call field 'setmode' (a nil value)
```

Also, the NodeMCU.bin built on linux is 619696 bytes while built on OS X is 477904. For both, it's a clean checkout of the dev-esp32 branch and no changes to menuconfig other than the serial port name. So we're onto something.... :)

bwhitman, 2017-01-18:
OK, I got it to compile on OS X with the modules. There's some issue with the system included version of make, so I installed GNU Make 4.2.1 via Homebrew and it works great now. after doing 

```
brew install homebrew/dupes/make
```

Just run gmake all or gmake flash.  To test the wifi module, I do:

```
wifi.start() (and wait)
wifi.mode(wifi.STATION)
wifi.sta.config({ssid="ssid",pwd="pw"})
```

luismfonseca, 2017-01-18:
Nice one @bwhitman, that also did work for me.

I'll write something up to document how to build the `dev-esp32` branch. I feel it will help OS X users a lot.

marcelstoer, 2017-02-23:
@luismfonseca can you please put that into a README for that branch?

LDmitryL, 2017-03-02:
I build NodeMCU for ESP-WROOM-32 to Msys2_Shell under Windows.
And also faced with the problem that some commands work and some don't.

`NodeMCU build ESP32 unspecified powered by Lua 5.1.4
> =node.heap()
206500

> =node.chipid()
stdin:1: attempt to call field 'chipid' (a nil value)
stack traceback:
stdin:1: in main chunk
> =node.flashid()
stdin:1: attempt to call field 'flashid' (a nil value)
stack traceback:
stdin:1: in main chunk
> =node.info()
stdin:1: attempt to call field 'info' (a nil value)
stack traceback:
stdin:1: in main chunk
>`

Is there a solution to this problem?

jmattsson, 2017-03-02:
Those functions are not available at this time, sorry. Last I looked Espressif wasn't providing access to those particular values. Maybe they're available by now? Whoever gets a chance could have a look at doing this under #1614.

Please be aware that for various reasons (lack of hardware availability being the biggest one), our ESP32 support is far from complete at this time. The documentation for the ESP32 firmware is available in the [docs/ directory](https://github.com/nodemcu/nodemcu-firmware/tree/dev-esp32/docs), and lists what is available so far. We also have [project](https://github.com/nodemcu/nodemcu-firmware/projects/1) tracking our ESP32 support.

@marcelstoer Do you think we could get our RTD config updated so we generated web docs for the `dev-esp32` branch too?

marcelstoer, 2017-03-02:
Like so https://nodemcu.readthedocs.io/en/dev-esp32/ you mean?

jmattsson, 2017-03-02:
Yes! Except now I see I've failed to update the `node` module docs in this branch :(

jmattsson, 2017-03-02:
FYI, tonight I dug through the latest esptool and worked out how it generates the chipid, and made a matching implementation for NodeMCU on the ESP32. Not sure when I'll have time to go through the rest of the `node` module docs and implement the missing bits and/or remove them from the doc file. Apologies, life has been rather busy and I haven't gotten any ESP32 play time in well over a month :/

devsaurus, 2017-03-17:
I also think we should put up a boiler plate README.md. Propose to start with the one in [`dev` branch](https://github.com/nodemcu/nodemcu-firmware/blob/dev/README.md), strip it down for yet unclear/unavailable topics and focus on the [chapter "Building the firmware"](http://nodemcu.readthedocs.io/en/dev-esp32/en/build/). Guess the whole git submodule and menuconfig thing fits in there.
I'd set up a first shot in a PR and hope for input from you guys.

Nerdvana2, 2018-01-28:
I've been happily flashing ESP8266 modules with esptool on a Mac, using builds from the (much appreciated) cloud build service.
After reading the above discussions I've used an Ubuntu VPS for ESP32 builds instead of locally on my Mac.  But I can't work out which binaries to move to the Mac, or if anything else is needed before flashing to the ESP32 with esptool. 
"NodeMCU.bin" seems an obvious candidate, but "esptool.py .... flash_write NodeMCU.bin" gives responses suggesting I've missed out some fundamental step(s)
Any and all pointers to appropriate docs or suggestions how to proceeed welcome!!

devsaurus, 2018-01-28:
From `make flash V=1`:
```
esptool.py --chip esp32 --port "/dev/ttyUSB0" --baud 921600 --before "default_reset" --after "hard_reset" write_flash -z --flash_mode "dio" --flash_freq "40m" --flash_size detect 0x1000 build/bootloader/bootloader.bin 0x10000 build/NodeMCU.bin 0x8000 build/partitions_singleapp.bin
```
The flash partitioning is described in http://esp-idf.readthedocs.io/en/latest/api-guides/partition-tables.html.

PR #1873 (2017-03-24): ESP32: Set up README.md and build/flash docs.
Fixes #1703.

This is a first shot at a README for the `dev-esp32` branch and providing basic directions for building the firmware.

Please feed back with additions or corrections to enable users with enough info to get started.