nodemcu/nodemcu-firmware #1705: http.get not working -> no dns resolution

markus507, 2017-01-06:
I'm fairly new to NodeMCU developing but I cannot get http.get to get work.

I've created an Image over  http://nodemcu-build.com with the following modules:
bit, dht, file, gpio, http, i2c, net, node, tmr, u8g, uart, wifi, tls

The NodeMCU is connected to WiFi. But when I execute the following Code:
```
print(wifi.sta.getip())

http.get("http://nodemcu-build.com", nil, function(code, data)
  if (code < 0) then
    print("HTTP request failed")
  else
    print(code, data)
  end
end)
```
I get the following Response:

```
dofile('http.lua')
192.168.1.128	255.255.255.0	192.168.1.11
> HTTP client: DNS failed for nodemcu-build.com
HTTP request failed

```
What's wrong?

Thank you
Markus

bakabird, 2017-01-06:
After the [issue\#1704](https://github.com/nodemcu/nodemcu-firmware/issues/1704) I face this problem now.
After finishing built and flash the new bin( own SSL ) into my chip.

I try this on my `init.lua`:

```lua 
cntURL("https://wishwall.wilddogio.com/-KZG4d57IkgO7pcwVtrW.json",2)
```
but i still failed and got that on my *Serial Tool*:

```
node.restart()
> 
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 27376, room 16 
tail 0
chksum 0xbe
load 0x3ffe8000, len 2412, room 8 
tail 4
chksum 0xe1
load 0x3ffe896c, len 136, room 4 
tail 4
chksum 0x50
csum 0x50
屸?俷?抮傭n|?llll`屸r抣宭?

NodeMCU custom build by frightanic.com
	branch: dev
	commit: 5425adefff62f9ea2094e3e4581a79f1424e4433
	SSL: true
	modules: file,gpio,http,mdns,net,node,pcm,tmr,uart,websocket,wifi,tls
 build 	built on: 2017-01-06 15:21
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
the index of the currently selected AP is: 1
Setting up WIFI...
> IP is 192.168.253.1
IP is 192.168.253.1
HTTP client: DNS failed for wishwall.wilddogio.com
HTTP request failed
```
SAD_D^D.

------------

then i try `net.dns.resolve()` by set this fun:

```lua
function resolve(url)
    net.dns.resolve(url, function(sk, ip)
    if (ip == nil) then print("DNS fail!") else print(ip) end
    end)
end
```

i got that now:

```
resolve("wishwall.wilddogio.com")
> DNS fail!
resolve("www.baidu.com") 
> DNS fail!
```

CONFUSED.  QAQ

djphoenix, 2017-01-06:
Are you tried to [set DNS servers](http://nodemcu.readthedocs.io/en/dev/en/modules/net/#netdnssetdnsserver) (e.g. for Google DNS - 8.8.8.8/8.8.4.4)?
Your example just works for me.

markus507, 2017-01-06:
Thanks djphoenix for your answer, but it's still not working:

Executed Code:

```
print(wifi.sta.getip())

net.dns.setdnsserver("8.8.4.4", 0)
print(net.dns.getdnsserver(0))

net.dns.resolve("www.google.com", function(sk, ip)
    if (ip == nil) then print("DNS fail!") else print(ip) end
end)
```
Result with debug Image:
```
dofile('http.lua')
192.168.1.128	255.255.255.0	192.168.1.11
8.8.4.4
net_create is called.
UDP server/socket is set.
> net_dns_found is called.
DNS Fail!
DNS fail!
net_dns_found is called.
DNS Fail!
net_delete is called.
DNS fail!
net_dns_found is called.
DNS Fail!
net_delete is called.
DNS fail!
```
Maybe http://nodemcu-build.com is not the right place to get an Image from?

Thanks 
Markus

marcelstoer, 2017-01-06:
> Maybe http://nodemcu-build.com is not the right place to get an Image from?

It's one of 3 ways to build NodeMCU, feel free to try the others: http://nodemcu.readthedocs.io/en/latest/en/build/

Both the cloud builder and DNS resolution work just fine on `master` & `dev`, I tested most of the code pasted here. We don't do support on this issue list though -> closing

_However_, I'm seeing lots of failures with the HTTP module on `dev` and I've been meaning to create an issue for a few days.

closed 2017-01-06 by marcelstoer

JarrodMP, 2017-03-26:
Hi Could you help me catch the error: -11 as I would like to print and instruction and then sleep the device
> dofile("Httptest.lua");
192.168.1.40
http send
nil	nil
> HTTP client: Disconnected with error: -11
HTTP client: Connection timeout
HTTP client: Connection timeout
>
Using lua code  
package.path = '*.lua;' .. package.path
require "config"

values = "80"
urlVal = "temperature"
SensorID =  node.chipid()

function httpsend(urlVal ,SensorID, values) 

 print(configvals.HOST);
--http.put("http://".. config.HOST .."/sensorlogs/PutDataHere.php/put/"..SensorID.."/".."sensors_temperature".."/"..values,

local msg = http.put("http://".. configvals.HOST .."/sensorlogs/PutDataHere.php/put/"..SensorID.."/".."sensors_"..urlVal.."/"..values,
  'Content-Type: text/plain\r\n',
  'sent',
          function(code, data)
            if (code < 0) then
              print("HTTP request failed")
              print(code)
              print("Go to sleep because of error")

                tmr.stop(6)
            
          tmr.alarm(6,restartdelay  * 1000,1,function() if (configvals.causedebugprints) then print (" Couldnt send page ") end
          --go_to_sleep(sleeptime) 
            end )

              
            else
              print(code, data)
              print(" in Else ")
              -- check data good stop timer 
              tmr.stop(5)
             
            end
          end) 
-- sleep 
print("http send")
print(code,data)
end

httpsend(urlVal ,SensorID, values)

marcelstoer, 2017-03-26:
@JarrodMP this closed, see #1010 for support