nodemcu/nodemcu-firmware #1706: Request: Clear Saved Wi-Fi config

jfollas, 2017-01-06:
### Clear Saved Wi-Fi config

### Justification
In an older version of NodeMCU, I would just do something like ```wifi.sta.config("","")``` and that would overwrite the saved config in flash. My understanding was that the SDK would not try to connect if the SSID was empty.

Changes to the wifi module now enforces a required SSID of at least one character:
https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/wifi.c#L676

### Workarounds
Set an invalid SSID, but SDK will try to connect to it: ```wifi.sta.config(" ", "")```. 

*Note: This also has a side effect of the SoftAP radio channel (since there's only one shared radio) changing constantly as the ESP tries to connect to a SSID that doesn't exist.*

devsaurus, 2017-01-06:
Have you already tried [`wifi.sta.autoconnect(0)`](http://nodemcu.readthedocs.io/en/dev/en/modules/wifi/#wifistaautoconnect)? IMO it's the correct solution for your use case: keep the wifi credentials in flash but avoid automatic connection. Plus it's a reversible switch.

jfollas, 2017-01-06:
Auto-connect is part of the battle that is easily solved. But thanks for reminding me of it.

Still looking for a way to clear the credentials saved (think about scenario where a device is intended to be transient and we need to set up a new network even if the old is still accessible, or is being passed off to a new owner, or when doing a factory reset, etc). 

I modified my forked code to get rid of the minimum length requirement and the old behavior still works (it doesn't try to connect if SSID is zero length). I may be an outlier, but this is the preferred solution for me.

marcelstoer, 2017-01-06:
> Clear Saved Wi-Fi config

I fully support that.

> In an older version of NodeMCU, I would just do something like ```wifi.sta.config("","")``` and that would overwrite the saved config in flash.

I wouldn't like to see this introduced again as it never felt like more than a cheap (maybe even unintended) workaround.

What did you have in mind, `wifi.reset()` which would also clear the mode or rather something like `wifi.sta.clearconfig()`?

jfollas, 2017-01-06:
> I wouldn't like to see this introduced again as it never felt like more than a cheap (maybe even unintended) workaround.

Perhaps, but having that ability on the ```.config()``` method would also retain backwards compatibility. :-)  (Just me being lazy about needing to update my scripts).

I don't see a SDK 2.0 method to do it, so the underlying C implementation would probably be the same anyways (set SSID and password to null strings).

The method name doesn't matter to me.  ```wifi.sta.clearconfig()``` may be the most self documenting name, though.

How does this also play into the fact that there are more than one wifi config slots now? (that was a weird one for me to discover in the newer SDK, too)

konradhuebner, 2017-01-08:
As I just stumbled upon exactly this question while coding, I fully support that request. The primary use-case for me is a reset-button on a device, which shall bring back "factory settings", and to me this means not trying to connect to a WiFi.

jfollas, 2017-01-25:
I spent some time on a wifi.sta.clearconfig() method, and will submit a PR soon.

While I was in the code, I realized that requiring a minimum password length of 8 characters is not something that is actually required by the 2.0 SDK. WPA itself has that requirement, but the ESP also supports connecting the ESP station to a WEP access point. _Note: They prevent WEP for the SoftAP, but they do permit it for the Station mode._

Is there any real benefit for validating the password/key string length when setting the station configuration? It's not like the ESP will connect if the provided key is wrong, and enforcing a rule like this for something out of our control may actually make NodeMCU incompatible with someone's [outdated] Wi-Fi network.

I'm proposing that we remove the minimum password requirement when setting the wifi.sta config because the minimum level of security used by that external device is not a concern of this firmware. If someone has an old network and needs to use a 5-character WEP IV to connect to it, then we should let them use it.

PR #1761 (2017-02-08): Feature/wifi clear config
Fixes #1706 .

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

Added `wifi.sta.clearconfig()` to remove the saved Wi-Fi configuration from flash.  Useful in a couple of scenarios:
- Factory Reset: To clear the saved configuration as a last step before giving [... cut]
closed 2017-02-08 by marcelstoer