nodemcu/nodemcu-firmware #1708: doco: rtcmem handling of values greater than 31 bits

eyaleb, 2017-01-07:
### Expected behavior
Returned value from `read32()` the same as written by `write32()`.
### Actual behavior
Returned value from `read32()` is a signed 32-bit integer.
### Test code
```Lua
> n=0x7fffffff rtcmem.write32(119,n) print(n, rtcmem.read32(119))
2147483647      2147483647
> n=0x80000000 rtcmem.write32(119,n) print(n, rtcmem.read32(119))
2147483648      -2147483648
> n=0xffffffff rtcmem.write32(119,n) print(n, rtcmem.read32(119))
4294967295      -1
> n=0x100000000 rtcmem.write32(119,n) print(n, rtcmem.read32(119))
4294967295      -1
> n=0x1000000000 rtcmem.write32(119,n) print(n, rtcmem.read32(119))
4294967295      -1
```
### NodeMCU version
Current `dev` but probably similar in other branches. Float build.
### Hardware
N/A
### Comment
I do not know if this is intended, in which case the doco should probably make it clear, or an error in crossing the signed/unsigned border (the SDK `rtc_mem` functions are unsigned).

It would be simpler to write the lower 32bits (truncated at the top and the bottom) and read them as the unsigned value they represent . Maybe offer `write32u()` and `read32u()` that write and read the lower significant 32 bits as unsigned?

And in general:
in our lua, values written as `0xn...` do not behave like decimally written integers do. Is this due to the fact that float is used to store integers which can be much larger than 32 bits?

NodeMCU:
```lua
> print(0xffffffff, 0x100000000, 0x1000000000)
4294967295      4294967295      4294967295
> print(4294967295, 4294967296, 68719476736)
4294967295      4294967296      68719476736
```
lua (5.1 and 5.3) on my PC handles numbers differently, same way for hex and decimal:
```lua
> print(0xffffffff, 0x100000000, 0x1000000000)
4294967295      4294967296      68719476736
> print(4294967295, 4294967296, 68719476736)
4294967295      4294967296      68719476736
```

pjsg, 2017-01-07:
This interesting... I think that the difficulty is having a reasonably compatible interface between the float and integer builds. 

In the integer builds, the values are in the range -2147483648 .. 2147483647 . There is a bigger range in the float build. 

The issues seem to be:

* What should happen (in the float build) if a value outside the range above is passed into write32? Maybe throwing an error is the right thing here.

* If so, then we would need the write32u to deal with the 0 .. 4294967295 range. (and a read32u to read them back).

If we took this approach, then I think that the read32u/write32u either don't appear on integer builds, or they throw errors if passed negative numbers (or would return a negative number).

Nothing very satisfactory....

eyaleb, 2017-01-08:
Or we could even have `write32u()` not care about the sign and simply store the 32 lower bits of the (2's complement) value. So -1 is stored as `0xffffffff`. Leave it to the user to not supply negatives or otherwise handle them:
```
n = rtcmem.read32u(addr)
if n > 0x7fffffff then n = 4294967296 - n end    -- note: the clearer 0x100000000 does not work
```
Yes, we still have the other problem, the unusual interpretation of large `0xn...` constants.

closed 2019-06-21 by stale[bot]