nodemcu/nodemcu-firmware #1711: Can't access REPL through USB (first access)

rsc1975, 2017-01-09:
### Expected behavior
From a terminal I try to access the Lua REPL with the command:
```bash
screen /dev/tty.SLAB_USBtoUART 115200  
```
### Actual behavior
I can access to the port but there is no prompt instead It appears "garbage" on the screen, just like:

```
nl`nr���n�`r��n�`���p�nn�llll...
```

### Test code
I already flashed and used the board with micropython (v1.8.6) without problems, so the board is Ok and the USB driver is installed (I can access the micropython REPL), the flashing method I think that It's also Ok, but maybe I'm missing something specific with nodemcu.

My flashing method is the following (on OSX):
```bash
$ python -m esptool --port /dev/cu.SLAB_USBtoUART erase_flash
esptool.py v1.2.1
Connecting...
Running Cesanta flasher stub...
Erasing flash (this may take a while)...
Erase took 44.5 seconds

$ python -m esptool --port /dev/cu.SLAB_USBtoUART -b 921600 write_flash -fm dio --verify -ff 40m -fs 32m 0x0 ~/Downloads/nodemcu-master-7-modules-2017-01-09-17-48-36-integer.bin
esptool.py v1.2.1
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0240
Writing 397312 @ 0x0... 397312 (100 %)
Wrote 397312 bytes at 0x0 in 5.9 seconds (543.2 kbit/s)...
Leaving...
Verifying just-written flash...
Verifying 0x606fb (395003) bytes @ 0x00000000 in flash against /Users/rob/Downloads/nodemcu-master-7-modules-2017-01-09-17-48-36-integer.bin...
-- verify OK (digest matched)
```
Once It is flashed I reset the board and try to access to the port with:

```bash
screen /dev/tty.SLAB_USBtoUART 115200  
```

I've also try to flash with the esptool provided by nodemcu repo, to use the 128M flash size of my board, but with the same result, I also tried floating/integer firmware, etc.

I don't know how to get further details, or if they are actually necessary, this problem It seems so basic that is probably known by the community, this is my first try with nodemcu.

### NodeMCU version
I've tried the "master" and "dev" versions provided by the cloud service: https://nodemcu-build.com/  with different modules selections.

### Hardware
I'm using a Wemos D1 mini PRO board

devsaurus, 2017-01-09:
Current `master` won't work properly with a Wemos D1 mini pro board, you need an image from `dev` (what you already tried apparently).
Please confirm whether the following setup works or not:
- Image built from current `dev`
- esptool.py from https://github.com/espressif/esptool
- Flash size parameter `-fs 8m` (while [16m and 32m are also confirmed to work](https://github.com/nodemcu/nodemcu-firmware/pull/1646#issuecomment-268593201))

We looked at pretty much the same setup like yours in the linked PR discussion.

marcoskirsch, 2017-01-09:
> Current master won't work properly with a Wemos D1 mini pro board, you need an image from dev (what you already tried apparently).

This is good to know. Tangentially related: Any idea when master will support Wemos D1

rsc1975, 2017-01-09:
**It works!!** (using 32m, I didn't try other options) thank you.

I did a lot of tests, and I'm not sure why didn't work the first time, maybe I used the custom esptool with 128m flash size, I don't know, I did a lot of tests and combinations, or maybe I forgot to erase the flash or who knows...

```bash
fs.start:400000,max:bfa000
mount res: 0, 0
Task task_lua started.
nul mode, fpm auto sleep set:enalbe

NodeMCU custom build by frightanic.com
        branch: dev
        commit: 378e5eb0ad04a555da4a9e4939d88fc771d1f13f
        SSL: false
        modules: cjson,cron,encoder,file,gpio,http,i2c,mqtt,net,node,rtcfifo,rtcmem,rtctime,spi,tmr,uart,websocket,wifi
 build  built on: 2017-01-09 17:56
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
lua: cannot open init.lua
> Heap size::41680.
```

The command that I finally used:

```bash
$ python -m esptool --port /dev/cu.SLAB_USBtoUART -b 921600 write_flash -fm dio --verify -ff 40m -fs 32m 0x0 ~/Downloads/nodemcu-dev-18-modules-2017-01-09-17-57-14-float.bin

```

closed 2017-01-09 by rsc1975

devsaurus, 2017-01-10:
@marcoskirsch next master drop which will include #1646 is planned to happen around Feb 1st.

marcelstoer, 2017-01-10:
@marcoskirsch you can track our release train at https://github.com/nodemcu/nodemcu-firmware/milestones (pity GH doesn't support Markdown in the descriptions)