nodemcu/nodemcu-firmware #1712: PCM Implementation panic

stonehippo, 2017-01-09:
### Expected behavior

When running the PCM example `play_file.lua`, the file `jump_8k.u8` is opened and played back via the sigma-delta PCM implementation.

### Actual behavior
The script exits with the following error:

```
PANIC: unprotected error in call to Lua API (bad argument #1 to '?' (file.obj expected, got userdata))
```

Checking the file SPIFFS system shows that the files for the example are in place.

```
filename: files.lua, size:82
filename: play_file.lua, size:887
filename: jump_8k.u8, size:6268
```

### Test code

See the example from https://github.com/nodemcu/nodemcu-firmware/tree/master/lua_examples/pcm/play_file.lua

### NodeMCU version

Here's the info for my custom NodeMCU build (created via the service):

```
NodeMCU custom build by frightanic.com
	branch: master
	commit: 81ec3665cb5fe68eb8596612485cc206b65659c9
	SSL: false
	modules: file,gpio,net,node,pcm,sigma_delta,tmr,uart,wifi
 build 	built on: 2017-01-09 04:29
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
```

### Hardware
I've tried running this on a plain ESP-01 module with 512K and an Adafruit ESP8266 HUZZAH.

devsaurus, 2017-01-10:
Thanks for reporting!
Please try this change:
```lua
drv:on("data", function(drv) file.read() end)
```

stonehippo, 2017-01-10:
@devsaurus Yes, that resolves it. The `play_file.lua` example works correctly when edited to pass a function to `drv:on("data")` rather than the results of `file.read`. I should have noticed that! :)

devsaurus, 2017-01-10:
Cool, I'll fix the example accordingly.

It got broken by the file object extensions because `file.read()` now investigates the parameter if it's user data and complains when it gets a pcm driver instead of a file handle.

PR #1720 (2017-01-11): Fix pcm examples to comply with current code base.
Fixes #1712.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

Fix pcm examples to comply with current code base.
closed 2017-01-12 by devsaurus