nodemcu/nodemcu-firmware #1718: enduser setup unstable / strange behaviour

markusf, 2017-01-11:
Hello guys,

I am experiencing certain strange issues with enduser setup:

1. sometimes when I run endusersetup, endusersetup seems to not go into manual mode and overrides my ssid ("TestSSID") with "SetupGadgetXXX"
2. when resetting my wifi credentials and restart enduser setup, enduser setup starts and sometimes automatically connects to the latest connected access point without touching the web interface
3. when connecting to the webinterface sometimes when scanning access points it returns no devices found (500 internal server error)
4. the status message below the html box is inconsistent (idle, connecting..)

Any ideas?

Best regards

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
local startSetup = function()
    print("start setup")
    wifi.setmode(wifi.STATIONAP)
    wifi.ap.config({ssid="TestSSID", auth=wifi.OPEN})
    enduser_setup.manual(true)
    enduser_setup.start(
      function()

        ssid, password, bssid_set, bssid=wifi.sta.getconfig()
        
        print("\nCurrent Station configuration:\nSSID : "..ssid
        .."\nPassword  : "..password
        .."\nBSSID_set  : "..bssid_set
        .."\nBSSID: "..bssid.."\n")

        tmr.alarm(6, 8000, tmr.ALARM_SINGLE, function()
            node.restart()
        end)
        
      end,
      function(err, str)
        print("enduser_setup: Err #" .. err .. ": " .. str)
      end
    );   
end
```
### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: 81ec3665cb5fe68eb8596612485cc206b65659c9
	SSL: true
	modules: cjson,crypto,enduser_setup,file,gpio,mqtt,net,node,pwm,rtcfifo,rtcmem,rtctime,tmr,uart,wifi
 build 	built on: 2016-12-11 05:09
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

### Hardware
ESP8266 latest version

markusf, 2017-01-11:
Seems like 3,4 have been fixed here: https://github.com/nodemcu/nodemcu-firmware/pull/1605
Will test the firmware tomorrow.

jfollas, 2017-01-26:
@markusf 

>sometimes when I run endusersetup, endusersetup seems to not go into manual mode and overrides my ssid ("TestSSID") with "SetupGadgetXXX"

I'm working on EUS this week, and will be concentrating on manual mode scenarios. I think bad behavior was not caught in the past because my primary scenarios were all automatic mode.

>when resetting my wifi credentials and restart enduser setup, enduser setup starts and sometimes automatically connects to the latest connected access point without touching the web interface

How are you resetting wifi credentials? I submitted a PR for the wifi module to help solve what seems to be a related scenario: when EUS starts and there's already a valid configuration, it will shut down. Recent changes to the wifi module made it impossible to clear the wifi credentials from flash, so my PR adds a method to do that. The new workflow (after the PR is merged), if you need to use EUS, will be to do `wifi.sta.clearconfig()` prior to `enduser_setup.start()`.

>when connecting to the webinterface sometimes when scanning access points it returns no devices found (500 internal server error)

Looking at the code, a 500 will be returned during the AP Scan if the ESP runs out of memory while trying to alloc space for the JSON string that it is building (approx 100 bytes needed per AP entry). I can see this as a problem if there are a lot of nearby networks, and I don't have a good solution other than to advise EUS be used early in the boot process when you have the most heap memory available. Ultimately, it will probably require someone with better C memory management skills to solve this, or maybe chunking the response.

>"connected" / success message, just showing when I manuall delay the restart (see timer snippt). According to the documentation it should wait 10 seconds.

> the status message below the html box is inconsistent (idle, connecting..)

I have spent time this week to try to resolve some of these status message issues. The complexity largely arises due to the fact that there is only one hardware radio on the ESP, and the Station takes priority over the SoftAP. So, when you use EUS to set configuration that requires the Station to change to a different channel, the SoftAP goes along with it. Your phone or laptop, however, doesn't know right away that the SoftAP has left the channel, so it's still trying to send/receive until a timeout occurs. 

If the Station can attach to the specified AP and get an IP address, then the code will shut down the station and change the channel back to the original one. If we're lucky, this happens quick enough and your phone/laptop just resumes where it left off.  

If we're unlucky, then the phone/laptop will begin to look for the SoftAP on different channels, and may find it on the new Station channel just prior to the successful connection when we change the SoftAP back to its original channel (but the phone/laptop is no longer there).  

If we're really unlucky, then all of this channel changing will cause the phone/laptop to give up and try to connect to one of its preferred access points (i.e., your home or office WiFi). Then the web page definitely won't receive a response for the status data.

I've discovered that it is a pretty complex dance of trying to sync up status polling on the EUS web page with whatever timing may be required for your ESP to connect to your WiFi network and obtain an IP address (which is our indication of success).

jfollas, 2017-02-08:
Here's an update on one finding as I worked to make the channel-switch handoff work better:

It seems that the Wi-Fi API in SDK 2.0 implements the Channel Switch Announcement in beacon frames (defined by 802.11h). That's a feature that was originally implemented because the 5 GHz Wi-Fi band shares radio spectrum with radar (i.e., airport radar, maybe doppler weather radar, etc). If the Access Point detects a radar being used on it's channel, then it needs to tell all of its connected clients to switch to another channel where they will not interfere. It does this by including a CSA in the beacon data or probe response.

This also seems to be a good solution in our case with the ESP down in the 2.4 GHz band, because if the SoftAP is running on some channel and the ESP's Station connects (causing the hardware radio channel to change), then the SoftAP can notify all of its connected STAs that it's time to move to that new channel.

In theory, of course. 

They have the CSA beacons implemented in the SDK, and with Wireshark, I can see that they indeed count down as they should (i.e., "we're going to change channels in 3 more beacons... 2 more... 1 more...").  But, I think there's a bug - after the last CSA beacon, the SoftAP is supposed to be on the new channel, but I see one more regular beacon being sent on the old channel. So, this may actually be resetting the channel change action of the STAs because they think the AP is still up on the old channel (just a theory that I have - kind of hard to prove).  

And, because there are no more beacons being sent on the old channel, the STA will eventually timeout and then the platform's rules decide what it should do next (e.g., try to find the SSID that it was just connected to, or try to connect to a better Wi-Fi in search of internet, etc).

The other thing that I think I discovered is that when you call `wifi.ap.config()` or `wifi_softap_set_config()`, even if it's just to change the channel, that the SoftAP will de-auth all of its connected STAs. This resets any 802.11 and TCP connections. So, there's no way to just change the channel of an already running SoftAP without effectively shutting it down and restarting it (which is what `.config()` seems to do).  Note: There is a `wifi_set_channel()` function intended for sniffer mode, but this seemingly has no effect when a SoftAP is already running.

So, even when we switch channels back to the old one after the Station connection has been confirmed and shut down, any STAs that were previously connected would have been de-authed, and would have to connect again.  And, in thinking about how Manual mode may be used, any connections to the SoftAP before EUS is started would be dropped if and when the SoftAP needs to change channels.

I'm awaiting a response to my [bug report](http://bbs.espressif.com/viewtopic.php?f=66&t=3328) about the extra beacon. I think that if that is solved, then the CSA mechanism itself may be sufficient without needing to switch the SoftAP back to its original channel. That would greatly simplify how EUS works.

sticilface, 2017-03-14:
I have also found this with extensive trial and error.  It makes for a very clunky configuration, as I have to wait for the user to reconnect to the AP to continue configuration after a mandatory channel switch.   Normally the client computer will reconnect to the normal WiFi, and the user is left wondering why the ESP8266 config is no longer working...  

So I would also appreciate some progress on this issue!

jmattsson, 2017-05-04:
@jfollas you might have more luck getting a response if you submit via the bug-bounty email?

igrr, 2017-05-04:
I have poked QA/engineers about this a few days ago, they are checking the issue.

jmattsson, 2017-05-04:
Thanks for that update @igrr!

jfollas, 2017-05-04:
I submitted to both the bugs forum and the bounty email. Have not heard anything myself.

closed 2019-06-21 by stale[bot]