nodemcu/nodemcu-firmware #1721: http.get() Memory Leak

zhujieo, 2017-01-12:
8<------------------------ BUG REPORT -----------------------------------------
My Egnlish is bad,Sorry!

I have problem:
    when Nodemcu connected my web server the memory is not leak,
    but when I close my web server, the nodemcu is disconnect the web service,it's memory is leak.

### Expected behavior

### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
wifi.setmode(wifi.STATION)
wifi.sta.config('****','*****')
wifi.sta.connect()

tmr.alarm(0,2000,1,function ()
    print(node.heap())
    http.get("http://192.168.1.149:8000/devices/command", nil, function(code, json)
        if (code < 0) then
            print('error')
        else
            print('ok')
        end
    end)
end)
```

ESPlorer Report:

NodeMCU 1.5.4.1 build 2016/12/31 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
> 40520
42848
42320
ok
42216
ok
42232
ok
42232
HTTP client: Connection timeout
ok
41464
HTTP client: Connection timeout
ok
41464
ok
41464
ok
41464
ok
41464
ok
41464
ok
41464
ok
41464
ok
41464
ok
41448 . -----------(This time I closed my server)
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
41264
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
40976
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
40728
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
40464
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
40184
HTTP client: Connection timeout
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
......

### NodeMCU version
Master

NodeMCU 1.5.4.1 build 2016/12/31 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

### Hardware
Nodemcu with nothing attached

8<------------------------ END BUG REPORT -------------------------------------

djphoenix, 2017-01-12:
Can you check that problem still appears on latest `dev`?

zhujieo, 2017-01-12:
OK, Let me try it. Thanks.

zhujieo, 2017-01-12:
In the latest dev ,this problem is still have.

`NodeMCU 2.0.0 build unspecified powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
> 40464
42712
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
42440
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
42192
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
41928
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
41680
HTTP client: Connection timeout
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
40672
HTTP client: Connection timeout
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
40432
HTTP client: Connection timeout
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
40176
HTTP client: Connection timeout
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
39936
HTTP client: Connection timeout
HTTP client: Disconnected with error: -9
HTTP client: Connection timeout
`

marcelstoer, 2017-01-12:
> when I close my web server, the nodemcu is disconnect the web service,it's memory is leak.

I vaguely remember having seen that claim somewhere else before...

zhujieo, 2017-01-13:
Thanks for you help.  @djphoenix  @marcelstoer

coderkevin, 2017-04-12:
I'm also experiencing this using just http.post() to a local IP address URL. I get about 70 "HTTP client: Connection timeout" lines before the unit crashes and restarts. As long as the HTTP requests complete without errors, everything is fine though. 🤔

felansu, 2017-04-24:
Thanks @marcelstoer for lock #1804 😡 

> Sending the Connection:close header with the request is a useful default behavior and that's now documented in our net.socket:on() example.

I'm thinking this connection would have to be closed by default. This is stupidly logical.

HHHartmann, 2017-07-01:
This could be fixed by issue #2007. Also note that "error" is never printed (and the cleanup is not executed)

marcelstoer, 2017-07-01:
PR fixes this, thanks a lot.

closed 2017-07-01 by marcelstoer

priyankakrvino, 2018-01-08:
srv = net.createServer(net.TCP)
srv:listen(80, function(conn)
    conn:on("receive", function(sck, payload)
        print(payload)
        sck:send("https://b7jgyq.messaging.internetofthings.ibmcloud.com:8883/api/v0002/device/types/esp8266/devices/rty567/events/eventId Content-Type: application/json, Authorization: Basic dXNlLXRva2VuLWF1dGg6RGZnaGo1Njc4","{"datetime":"2017-11-14T14:08:23.498Z", "temp":24}")
    end)
    conn:on("sent", function(sck) sck:close() end)
end)

the error of the above code is HTTP client: Disconnected with error: -9
HTTP client: Connection timeout

in node mcu http and mqtt both couldn't work
is there any solution for that?????

HHHartmann, 2018-01-09:
This issue is closed and fixed. Please pen a new bug or see the documentation for support sources. 

I assume you call the above server with http.Get from elsewhere? 

Http expects a valid http response which your code does not supply so it throws an error

priyankakrvino, 2018-01-17:
station_cfg={}
station_cfg.ssid="T"
station_cfg.pwd="123"
wifi.sta.config(station_cfg)
 
function postThingSpeak(level)
    connout = nil
    connout = net.createConnection(net.TCP, 0)
 
    connout:on("receive", function(connout, payloadout)
        if (string.find(payloadout, "Status: 200 OK") ~= nil) then
            print("Posted OK");
        end
    end)
 
    connout:on("connection", function(connout, payloadout)
 
        print ("Posting...");
 
        local volt = node.readvdd33();      
 
        connout:send("GET /update?api_key=fwhg69237612&field1=" .. (volt/1000) .. "." .. (volt%1000)
        .. " HTTP/1.1\r\n"
        .. "Host: api.thingspeak.com\r\n"
        .. "Connection: close\r\n"
        .. "Accept: */*\r\n"
        .. "User-Agent: Mozilla/4.0 (compatible; esp8266 Lua; Windows NT 5.1)\r\n"
        .. "\r\n")
    end)
 
    connout:on("disconnection", function(connout, payloadout)
        connout:close();
        collectgarbage();
    end)
 
    connout:connect(80,'api.thingspeak.com')
end
 
tmr.alarm(1, 60000, 1, function() postThingSpeak(0) end)

for this code also creating some error like below

PANIC: unprotected error in call to Lua API (init.lua:20: attempt to call field 'readvdd33' (a nil value))

how could i solve this error???

marcelstoer, 2018-01-17:
Yours is a question and this issue is closed. See #1010.