nodemcu/nodemcu-firmware #1723: MDNS query

reejinbouk, 2017-01-14:
nodemcu firmware stil can't connect to devices with avahi-daemon service.
for example it can't connect to "raspberrypi.local" 
propose need to add port 5353 to "net" module, then connect to multicast, set dnsserver and it will work?

djphoenix, 2017-01-19:
Not sure this may be intergated in `net` module (or elsewhere) because MDNS is not part of LWIP.
But I think that MDNS resolution may (and should) be implemented in `mdns` module.
Tagging @pjsg as maintainer of `mdns` module

pjsg, 2017-01-22:
I don't have any experience with using nodemcu as a mdns client. The approach outlined by @reejinbouk ought to work.

dwery, 2017-01-22:
@pjsg @djphoenix just found this: http://desperate-programmers.com/2016/02/02/an-update-to-mdns-querries-yay/

the .c file provides the parsing routines, while the lua code has the high level stuff

iShift, 2017-02-21:
+1 add mDNS resolving function to firmware

closed 2019-06-21 by stale[bot]