nodemcu/nodemcu-firmware #1726: Bug in wifi.c, line 740

neilyoung, 2017-01-14:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

### Expected behavior
Wifi password of 64 byte length is accepted
### Actual behavior
Wifi password of length 64 is rejected in line 740, because the comparison MUST be done with '<=' instead of '<' only. Corrected code:

``` 
luaL_argcheck(L, (pl==0||(pl>=8 && pl<=sizeof(sta_conf.password)) ), 2, "length:0 or 8-64");
```
### Test code
```Lua
wifi.sta.config("SSID", "0123456789012345678901234567890123456789012345678001234567890123")

```
This will be rejected in line 740, although valid.

### NodeMCU version
master, 1.5.4.1

### Hardware
NodeMCUv2

neilyoung, 2017-01-16:
Hmm.. Nobody? I thought it would be an easy issue...

neilyoung, 2017-01-16:
Thanks for fixing it

closed 2017-01-16 by neilyoung

marcelstoer, 2017-01-17:
> I thought it would be an easy issue...

It is...and we do accept PRs 😉 

\<banging-the-marketing-drum\>
Building NodeMCU on your own machine is actually [fairly easy using Docker](http://nodemcu.readthedocs.io/en/latest/en/build/#docker-image) - regardless of OS.
\</banging-the-marketing-drum\>

PR #1730 (2017-01-17): Wifi password of 64 byte length is accepted
Fixes #1726 .

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [X] The code changes are reflected in the documentation at `docs/en/*`.

Allows usage of a 64 character long WiFi password as reported by @neilyoung.
Could not test it though.