nodemcu/nodemcu-firmware #1731: mqtt: event 'offline' sometimes not called

JosefRypacek, 2017-01-17:
### Expected behavior

### Actual behavior
Hi,
I have some ESP devices and they publish every 5 or 60 seconds values over mqtt. I'm using two builds (see section below).
**After 1 hour or few days mqtt becomes unresponsible. Broker gets offline message, but ESP will not get "offline" event.**
ESP client writes just: _mqtt: connection failed reason: -5_ (my debug message). It's from "callback function for when the connection could not be established". Event "offline" is not called and auto-reconnect (is set as you can see) does not work. 
If I try to reproduce this (stop broker etc...) only event "offline" is called and everything is works fine. 
Also if server is stopped when starting ESP, then ESP writes the same message (mqtt: connection failed reason), but auto-reconnect works fine here.

Am I doing something wrong? Is it a good idea to set both "callback function for when the connection could not be established" and "offline event"? I think it's a bug somewhere in mqtt client.

### Test code
There are most important commands of my code
```Lua
  m:lwt(BASETOPIC, 'offline', 1, 1)

  m:connect(HOST, PORT, 0, 1, nil,
  function(client, reason)
    print('mqtt: connection failed reason: ' .. reason)
  end)

  m:on('offline', function(client)
    print('offline at: ' .. tmr.time())
  end)
```
### NodeMCU version
I'm using following builds on my devices:
- nodemcu-dev from 2016-11-17-14-08-47
- nodemcu-dev from 2017-01-05-14-26-20

pjsg, 2017-01-24:
If auto-reconnect is set, when do you think the "offline" event should be triggered?

There is a PR #1683 which tries to improve the reliability of mqtt. One of the big things is to deprecate the use of `autoreconnect` in favor of explicit retries (from the error methods). This
gives the application author the most control over retries.

NicolSpies, 2017-01-24:
@pjsg Way back when I started reporting about the mqtt reconnect, the following would have been nice.

- If auto-reconnect is set, when the mqtt connection becomes "stale" or unresponsive, auto-reconnect is tried as many times as I have told it to try to reconnect. If it is does not succeed after the tries the "offline" event in triggered with a auto-reconnect "true" flag allowing me to "know" reconnect failed and will allow for specific code in the callback.
- If auto-reconnect is not set, the "offline" event is triggered immediately with a auto-reconnect "false" flag.

This way the developer can let mqtt try to reconnect first before passing the "offline" callback.

**Now**, different scenarios I experienced complicates the suggestion above. The scenarios consists of a combination of one or more of the following:

1. ESP connection to AP  failed.
2. ESP connected to AP but AP connection to the internet failed.
3. ESP connected to AP and AP connected to internet but connection to the mqtt broker failed.

Dependent on the existence and combination of the above, auto-reconnection is futile and hence my original response regarding the "ping" functionality to establish what has failed before attempting an reconnect. 

**Thus as a final thought** regarding the auto-reconnect: It should only be attempted if none of the 3 conditions above are true or present. If this can be incorporated in the back-end functionality and returned as flags in the callback it will be great.

JosefRypacek, 2017-01-24:
I like auto-reconnect because I need to have mqtt connection always established. I can turn it off and wrote my own reconnect but... why?

This issue is not about auto-reconnect. Offline event is called every time connection (established connection) failed. Then auto-reconnect will connect again in few seconds. But sometimes connection (established connection) fails, offline event is not called and auto-reconnect will not establish the connection again. Instead the second callback is called (reason -5) and I need to reset device to connect again.

NicolSpies, 2017-01-24:
Yes, I agree 100%, this is exactly what I reported in #1406 last year

closed 2019-06-21 by stale[bot]