nodemcu/nodemcu-firmware #1734: bme280 reports wrong temperature around/below 0

dicer, 2017-01-17:
### Expected behavior
Module returns actual temperature when reading it.

### Actual behavior
Around 0 degrees Celsius the temperature I read has 1-2° jumps in it and generally looks wrong:
[image]

I don't know the actual temperatures but I would guess something goes wrong when going below 0°. It already starts with the jump from 0 to 1 degree. I'm pretty sure this should not happen and the temperature was already below 0, before it jumped to -2°
I think the graph should have shown the temperature sink below 0 to roughly -2 and then rise again above 0.

I've never seen these jumps before when measuring higher temperatures! So I don't think it's my backend code.

### Test code
Nothing special here (I basically follow the example from the docs):

bmetype = bme280.init(PIN_BME280_SDA, PIN_BME280_SCL)
P = bme280.baro()
P = string.format("%d.%03d", P/10, P%10)
H, T = bme280.humi()
T = string.format("%d.%02d", T/100, T%100)

### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: 8e48483c825dea9c12b37a4db3d034fccbcba0bf
	SSL: true
	modules: bme280,cjson,dht,file,gpio,http,i2c,mqtt,net,node,spi,tmr,uart,wifi,ws2812,file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2016-09-16 13:13
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

I know, rather old. But looking at bme280.c nothing changed since Aug 2, 2016

### Hardware
BME type: 2

devsaurus, 2017-01-17:
There are some improvements in `dev` branch that aren't available yet in `master`: #1652 and #1662
Please cross-check your application with a build from current `dev` branch.
Also note the updates to the API.

FrankX0, 2017-01-17:
You are running in unexpected (but correct) result of the modulo operation on negative numbers.
For example check [Modulo operation with negative numbers](http://stackoverflow.com/questions/11720656/modulo-operation-with-negative-numbers)

The example is correct for positive numbers, but incorrect for negative ones (and should be corrected).
Try something like:
```
if T<0 then
    str = "-"..(-T/100).."."..(-T%100)
else
    str = (T/100).."."..(T%100)
end
```

dicer, 2017-01-17:
Oh yeah, you are right @FrankX0 !
    T = -010
    T = string.format("%d.%02d", T/100, T%100)
    print (T)

-> 0.9
T = -110 -> -1.90

The following works:
T = -010
if T<0 then
    T = string.format("-%d.%02d", -T/100, -T%100)
else
    T = string.format("%d.%02d", T/100, T%100)
end
print (T)

I updated the doc in this pull request: https://github.com/nodemcu/nodemcu-firmware/pull/1735

closed 2017-01-17 by marcelstoer

vsky279, 2017-01-19:
Disadvantage of programming in summer... Thanks for spotting it. 

I'd prefer this syntax to fix the issue:
```lua
T = -010
local sgn = T < 0 and -1 or 1
local Ta = sgn * T
print(string.format("%s%d.%02d", sgn < 0 and "-" or "", Ta/100, Ta%100))
```

marcelstoer, 2017-01-19:
Same goes for examples in module documentation: http://nodemcu.readthedocs.io/en/latest/en/modules/bme280/

vsky279, 2017-01-19:
I will prepare a PR.

PR #1735 (2017-01-17): bme280: Correct code samples for negative values
Fixes #1734.

- [x] This PR is for the dev branch rather than for master.
- [x] This PR is compliant with the other contributing guidelines as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at docs/en/*.

The formatting/modulo will produce wrong values for negative numbers. Corrected the samples to make it work
PR #1794 (2017-02-12): BME280: Correct code samples for negative values - more elegant solution
Fixes #1734.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This PR modifie [... cut]