nodemcu/nodemcu-firmware #1736: ws2812:buffer:mix does not mix correctly due to rounding error

juanelc, 2017-01-17:
### Expected behavior
When 1/8 of 10 22 54 is mixed with 7/8 of 10 27 55 the result should be 10 23 54
### Actual behavior
When 1/8 of 10 22 54 is mixed with 7/8 of 10 27 55 the result is 9 22 53
### Test code
```Lua
buffer1 = ws2812.newBuffer(1, 3) 
buffer2 = ws2812.newBuffer(1, 3) 
buffer1:fill(10,22,54)
buffer2:fill(10,27,55)
buffer1:mix(256/8*7,buffer1,256/8,buffer2)
print(buffer1:get(1))
```
old result: 9 22 53
expected result: 10 23 54

I think it is because the shift operation is done before the value is summed.
line 402:     
```Lua
val += ((int)(source[src].values[i] * source[src].factor) >> 8);
```
I propose to remove the shift operation in line 402 and add it after the for loop. 
line 404:     
```Lua
val = val >> 8
```

### NodeMCU version

- nodemcu-master-9-modules-2017-01-13-07-19-37-float

### Hardware
ESP-01, 1MByte, WS2812

pjsg, 2017-01-17:
That sounds like a good solution. Please submit a PR to that effect.

HHHartmann, 2019-06-25:
This shouldn't get lost

marcelstoer, 2019-06-25:
Thanks Gregor

pjsg, 2019-06-25:
The code now does actually have the >> outside the loop (as of Oct 2017). Can you check to see if your problem is resolved?

HHHartmann, 2019-07-14:
The current code returns 
current result: 10 22 54   which still differs from
expected result: 10 23 54
The exact result is 10 22.625 54.125
so this seems to be a rounding issue.

HHHartmann, 2019-09-22:
fixed in  releases 3.0-master_20190907

PR #2832 (2019-07-14): fix rounding in ws2812:buffer:mix
Fixes #1736.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
closed 2019-09-22 by HHHartmann