nodemcu/nodemcu-firmware #1737: Is there a race in eventmon?

eyaleb, 2017-01-18:
This is not a simple missed trigger bug but a design (SDK or nodemcu?) issue.
### Expected behaviour
Registering an event should inform of the event happening.
### Actual behaviour
The event does not always trigger.
### Test code
If I do this early (to stop dhcp)
```lua
wifi.sta.setip({ip="...",netmask="...",gateway="..."})
```
then (at times much) later I attend to the wifi
```lua
eventmon.register(eventmon.STA_CONNECTED, function(T)
    print("status="..wifi.sta.status()
end)
```
I usually see
```
status=5
```
However, if the registration is too late, after the connection is established, then no trigger is generated and wdt reset takes place.

I do not see a simple way to ensure I **always** get notified. I now do this:
```lua
wifi.sta.setip({ip="...",netmask="...",gateway="..."})
...
local handled = false
eventmon.register(eventmon.STA_CONNECTED, function(T)
    if not handled then
        handled = true
        print("status="..wifi.sta.status())
        whatever...
    end
end)
if not handled and 5 == wifi.sta.status() then
    handled = true
    print("handling missed event")
    whatever...
end
```
but is setting `handled` properly serialised?
### NodeMCU version
Latest `dev`
### Hardware
Any

djphoenix, 2017-01-18:
Lua is event-based and have only one "thread". So lua callbacks may not ever be called simultaneously.

eyaleb, 2017-01-18:
Thanks @djphoenix I now re-read the [lua doc](http://nodemcu.readthedocs.io/en/dev/en/lua-developer-faq/#how-is-nodemcu-lua-different-to-standard-lua) and can see how my code is safe, and does not even need the protection of `handled`.

However, the `5 == ...status` section is still required. It would be more reliable if the `CONNECTED` (and `GOT_IP`) could fire when the status is already satisfied at the time of registration. This will allow registering the event without worrying about the timing.

Should our doco (and examples) clarify this? Or is my understanding still lacking?

dnc40085, 2017-01-19:
@eyaleb as long as the event callback is registered before the event happens, it should always fire the callback.

The event handling for `wifi.eventmon` is as follows:
* The SDK calls event handler callback (in c)
* check if user has registered an event callback (in Lua). if not, return and discard event.
* if the user has registered an event callback (in Lua), add the event to the end of the event queue
* Post a task to process the event queue at a later time
* When the task executes, arrange the relevant event data in to a Lua table and return it to the user's event callback
* remove event from queue. if event queue is not empty, post the task again

eyaleb, 2017-01-19:
@dnc40085 I think I now understand this process, thanks. Related though: if I unregister the event, will any corresponding queued task be removed? I expect so.

My main point remains: registering a CONNECT event does not guarantee I will know that a connection was made. I must check the status too. In short, eventmon by itself is not enough, there is a timing issue (maybe 'race' in the title is inaccurate).

dnc40085, 2017-01-19:
>registering a CONNECT event does not guarantee I will know that a connection was made. I must check the status too.

I believe the `wifi.eventmon.STA_CONNECTED` event is indicating that the station has successfully completed the network authentication and the `wifi.eventmon.STA_GOT_IP` event indicates that the station now has an address and is ready to communicate with the network.

eyaleb, 2017-01-19:
@dnc40085 You are correct, however for this issue it does not matter if I use `STA_CONNECTED` or `STA_GOT_IP`.  The point is that whichever event you register, you cannot trust it to fire unless you register **before** the event actually happened, a point in time that is not known to your app. It is a crapshoot.

BTW, I tested, and the status is already 5 (`wifi.sta.STA_GOTIP`) when I enter the `STA_CONNECTED` event. Not that it matters for myapp.

marcelstoer, 2017-01-19:
Can be considered resolved I guess.
With an event driven system you always need to register the callbacks before the function that triggers those events is invoked.

It's just a bit unfortunate that we've got both `wifi.sta.eventMonReg()` and `wifi.eventmon.register()`. Our documentation states:

> Note: The functions wifi.sta.eventMon___() and wifi.eventmon.___() are completely seperate and can be used independently of one another.

but w/o giving recommendations as for when to use what that isn't really helpful.

PR #1889 (2017-03-31): Update wifi event monitor documentation
Update wifi event monitor documentation.

Fixes #1737.

- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] This PR is for the `dev` branch rather than for `master`.

A lack of information about when the wifi event monitor callbacks should be registered is causing confusion, the new information provided should help to clarify the function of the event monitor .
closed 2017-01-19 by marcelstoer