nodemcu/nodemcu-firmware #1739: Extend node.info([buildRef,[modules]])

spaccabit, 2017-01-18:
Hi at all,
I'm trying different builds with different settings to understand how nodeMcu.
The more difficulty that I met it is to identify the build of firmware and the selected modules.

I would suggest adding to node.info optional parameters:

buildRef - returns the short hash of the commit to which the build refers
modules - returns a table containing the list of modules selected on user_modules.h

djphoenix, 2017-01-19:
For cloud builds there are some information that printed on bootup.
Not sure how it works at all. @marcelstoer I think you may describe it?

So I agree for expose that data somewhere in Lua-level.

marcelstoer, 2017-01-19:
My cloud builder simply redefines `NODE_VERSION` and `BUILD_DATE` in `user_version.h`. The Git revision is extracted by running `git rev-parse HEAD`.

There are more details here https://github.com/nodemcu/nodemcu-firmware/issues/449#issuecomment-109283562.

HHHartmann, 2019-07-05:
I am working at #2799 and would like to add these informations too. 
The idea is to return all of them in node.info.

HHHartmann, 2019-09-20:
fixed by #2830

closed 2019-09-20 by HHHartmann