nodemcu/nodemcu-firmware #1740: How many requests nodemcu can handle at a time

Akriti-S, 2017-01-19:
If we have 4 connections to a nodemcu (esp 8266 nodemcu 12E board) and all of the connections are sending voluminous traffic to it, then in this case will it be able to handle such condition.

marcelstoer, 2017-01-19:
A related comment: https://github.com/nodemcu/nodemcu-firmware/issues/1737#issuecomment-273480908

Akriti-S, 2017-01-24:
Thank you for your response. My query is bit different, I want to know about the ideal load of traffic(number of requests from clients) a nodemcu web server can handle at a time.

marcelstoer, 2017-01-29:
If a system is single-threaded: 1 concurrent request. Also see https://github.com/nodemcu/nodemcu-firmware/pull/1757#issuecomment-275877454

closed 2017-01-29 by marcelstoer