nodemcu/nodemcu-firmware #1747: gpio,tmr didn't work in ESP32 lua nodemcu-firmware

kevinlovejia, 2017-01-20:
### Expected behavior
same code worked in ESP8266 module
### Actual behavior
Error 1
stdin:1: attempt to index global 'gpio' (a nil value)gpio.write(3,gpio.HIGH)
Error 2
bh1750_Example1.lua:9: attempt to index global 'tmr' (a nil value)
### Test code
```Lua
for Error 1
gpio.mode(3,gpio.OUTPUT)
gpio.write(3,gpio.HIGH)

for Error 2
tmr.alarm(0, 10000, 1, function()

    SDA_PIN = 6 -- sda pin, GPIO12
    SCL_PIN = 5 -- scl pin, GPIO14

    bh1750 = require("bh1750")
    bh1750.init(SDA_PIN, SCL_PIN)
    bh1750.read(OSS)
    l = bh1750.getlux()
    print("lux: "..(l / 100).."."..(l % 100).." lx")

    -- release module
    bh1750 = nil
    package.loaded["bh1750"]=nil

end)
```
### NodeMCU version
firmware is dev-esp32
I use the tool 'FLASH_DOWNLOAD_TOOLS_V3.4.4' download the 'Ai-Thinker_NodeMCU-32S_DIO_32Mbit_V1.0_20161101.bin' into ESP32S module.
### Hardware
ESP32S

zelll, 2017-01-21:
You got a 32S, then you must know ESP32 is a very new chip. A firmware to be ready for all features in a month? _Too young too naive ..._
Please download dev-esp32 branch and compile if you want to follow the development, though `tmr` and `i2c` are not provided at this moment.

kevinlovejia, 2017-01-21:
@zelll TKS, I've followed the dev-esp32 branch. Are u the firmware developer ? How much the development work was done?

zelll, 2017-01-21:
I'm not.
You can search issues with keyword "ESP32". There are many modules to be implemented. Today Espressif released IDF-2.0RC1, in which I2C/SPI/I2S drivers were added. Many developers are not woking for $, so we should waiting... And waiting for chip supply...

kevinlovejia, 2017-01-21:
Got it. Thanks ur reply.

devsaurus, 2017-01-21:
> You can search issues with keyword "ESP32".

We've grouped the ESP32 specific workpackages in a [github project](https://github.com/nodemcu/nodemcu-firmware/projects/1).

@zelll you'll figure that many basic modules are still waiting to be ported. `tmr` and `i2c` are two that are required by your scripts 😞

LDmitryL, 2017-03-03:
А есть прогнозы когда будет готов модуль TMR?

devsaurus, 2017-03-03:
We don't have specific estimations when the tmr module will be available.
@jmattsson and myself are currently the only active devs working on the esp32 branch and we both can't spend much time at the moment.
Personally I keep esp32 support on a low base level between more urgent topics for the esp8266.

devsaurus, 2017-03-08:
`tmr` module is now in `dev-esp32` branch, though without `tmr.delay()` and friends.

LDmitryL, 2017-03-09:
Launched here's an example:
`
mytimer = tmr.create();  
mytimer:register(10000, tmr.ALARM_AUTO, function() print("hey there") end);  
mytimer:interval(1000); 
mytimer:start(); 
 `
get a response every 10 seconds - `hey there`
that is, the operator `mytimer:interval(1000)` did not work?
Start again without operator `interval`. Get a response every 10 seconds - `hey there`
send command `mytimer:interval(1000)`
Print the message `hey there` terminated.
This is the correct behavior of this command?

djphoenix, 2017-03-09:
@LDmitryL nope I think. I tested original behaviour when created OO-API for `tmr`. `:interval()` should change an interval, keeping timer state as-is.

LDmitryL, 2017-03-09:
But what is interesting if to send a command
`mytimer:interval(100);
mytimer:stop();
mytimer:start();
`
Print `hey there` will change the interval of about 1 second.
A `mytimer:interval(500); mytimer:stop(); mytimer:start();`
the interval will be 5 seconds. Documentation the interval is set in miliseconds.
That is 500/1000=0.5 seconds. And I have 5 seconds. What is not taken into account when you run the example?
And most importantly - you have to do after you set the interval to do stop and then start the timer.

devsaurus, 2017-03-09:
@LDmitryL interesting observations, indeed. I'd guess that the FreeRTOS timers have specific constraints for changing the interval. Could require some more glue logic in the Lua layer to work around that.
I'll have a look.

djphoenix, 2017-03-09:
@devsaurus for RTOS timers you specify interval in "ticks" that may be accessed in portTICK_RATE_MS (if I know it right). Also tick rate is configurable via kconfig (for IDF).

devsaurus, 2017-03-09:
Well, I missed the `pdMS_TO_TICKS` conversion in the interval function (it's in place for the register)...

devsaurus, 2017-03-09:
a3dc13e3fb19c79282a1030bb3997aaafaeaf7a4 fixes `tmr.interval()`:
- apply correct period
- a running timer can be re-configured on the fly

It should expose the documented functionality now - no special handling required for changing the interval.

kevinlovejia, 2017-04-18:
@zelll @devsaurus ,How can I compile the latest ESP32 lua firmware? Do you have any tutorial about it? I git clone the latest dev-esp32 code, I used msys2_shell.cmd run "make menuconfig" is ok, when run "make flash" I met a problem ,see below.
/bin/sh: python: 未找到命令（can't find the command）

reference web:
http://wiki.ai-thinker.com/esp32/examples/nodemcu?s[]=esp32&s[]=idf&s[]=menuconfig
http://esp-idf.readthedocs.io/en/latest/linux-setup.html#step-1-download-binary-toolchain-for-the-esp32
http://esp-idf.readthedocs.io/en/latest/windows-setup.html

devsaurus, 2017-04-18:
> /bin/sh: python: 未找到命令（can't find the command）

@kevinlovejia I don't run Windows, but from the error message I guess that something with the python installation is not ok.

zelll, 2017-04-19:
@kevinlovejia 
download https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20170330.zip , unzip,
open mingw32.exe, run
```
git clone https://github.com/nodemcu/nodemcu-firmware.git
cd nodemcu-firmware
git checkout dev-esp32
git submodule update --init
make menuconfig
make
```
You should get a ```NodeMCU.bin``` in build/.

kevinlovejia, 2017-04-21:
@zelll I followed your steps and got the NodeMCU.bin finally. Thank you very much.

closed 2017-04-24 by kevinlovejia