nodemcu/nodemcu-firmware #1749: mbedtls cipher suites are poorly documented

nwf, 2017-01-20:
### Expected behavior

http://nodemcu.readthedocs.io/en/dev/en/modules/tls/ links to https://tls.mbed.org/core-features which reads as if ECDSA-based cipher suites are included by default.

### Actual behavior

The mbedTLS linked in to nodemcu supports only RSA-based cipher suites.  Disappointing, but I suppose this is a feature of the SDK and not something that nodemcu can change easily.  The docs could at least give warning.

### Test code
Running on 192.168.1.2 `openssl s_server -tls1_2 -key ec.key -cert ec.cert`, where those are EC keys, and then

```Lua
srv = tls.createConnection()
srv:connect(4433,"192.168.1.2")
```

will throw `:error:1417A0C1:SSL routines:tls_post_process_client_hello:no shared cipher:ssl/statem/statem_srvr.c:1422`.

### NodeMCU version

dev branch; commit a271ce4d34da4efb2a25114da7a5725d344c85ce

### Hardware

Standard breakout board, ESP-12E, I think.

marcelstoer, 2017-01-20:
> mbedTLS linked in to nodemcu supports only RSA-based cipher suites.

Happy to update the docs if you, or anyone else (@djphoenix?), can point to references for that in code or Espressif/SDK resources.

nwf, 2017-01-21:
The best I have been able to find, other than running the wireshark ssl dissector on the handshake, is in the *rtos* SDK: https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/include/mbedtls/config_esp.h#L649 and https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/include/mbedtls/config_esp.h#L673 .

djphoenix, 2017-01-21:
We use opensource version of mbedTLS, and (in #1435) we have added an [`user_mbedtls.h`](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/include/user_mbedtls.h) config file.
So, yes, ECDSA is disabled by default, but it may be enabled by one-two changes. I think we should document it in `tls` module docs.
And so, @marcelstoer, can we add some configuration tool in cloud builder?

marcelstoer, 2017-01-21:
Thanks Yury for the reference to `user_mbedtls.h`. I'll update the documentation.

I had my firmware send a request to https://www.howsmyssl.com/a/check. The result:

```json
{
  "given_cipher_suites": [
    "TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384",
    "TLS_DHE_RSA_WITH_AES_256_GCM_SHA384",
    "TLS_DHE_RSA_WITH_AES_256_CCM",
    "TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384",
    "TLS_DHE_RSA_WITH_AES_256_CBC_SHA256",
    "TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA",
    "TLS_DHE_RSA_WITH_AES_256_CBC_SHA",
    "TLS_DHE_RSA_WITH_AES_256_CCM_8",
    "TLS_ECDHE_RSA_WITH_CAMELLIA_256_GCM_SHA384",
    "TLS_DHE_RSA_WITH_CAMELLIA_256_GCM_SHA384",
    "TLS_ECDHE_RSA_WITH_CAMELLIA_256_CBC_SHA384",
    "TLS_DHE_RSA_WITH_CAMELLIA_256_CBC_SHA256",
    "TLS_DHE_RSA_WITH_CAMELLIA_256_CBC_SHA",
    "TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256",
    "TLS_DHE_RSA_WITH_AES_128_GCM_SHA256",
    "TLS_DHE_RSA_WITH_AES_128_CCM",
    "TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256",
    "TLS_DHE_RSA_WITH_AES_128_CBC_SHA256",
    "TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA",
    "TLS_DHE_RSA_WITH_AES_128_CBC_SHA",
    "TLS_DHE_RSA_WITH_AES_128_CCM_8",
    "TLS_ECDHE_RSA_WITH_CAMELLIA_128_GCM_SHA256",
    "TLS_DHE_RSA_WITH_CAMELLIA_128_GCM_SHA256",
    "TLS_ECDHE_RSA_WITH_CAMELLIA_128_CBC_SHA256",
    "TLS_DHE_RSA_WITH_CAMELLIA_128_CBC_SHA256",
    "TLS_DHE_RSA_WITH_CAMELLIA_128_CBC_SHA",
    "TLS_ECDHE_RSA_WITH_3DES_EDE_CBC_SHA",
    "TLS_DHE_RSA_WITH_3DES_EDE_CBC_SHA",
    "TLS_RSA_WITH_AES_256_GCM_SHA384",
    "TLS_RSA_WITH_AES_256_CCM",
    "TLS_RSA_WITH_AES_256_CBC_SHA256",
    "TLS_RSA_WITH_AES_256_CBC_SHA",
    "TLS_ECDH_RSA_WITH_AES_256_GCM_SHA384",
    "TLS_ECDH_RSA_WITH_AES_256_CBC_SHA384",
    "TLS_ECDH_RSA_WITH_AES_256_CBC_SHA",
    "TLS_RSA_WITH_AES_256_CCM_8",
    "TLS_RSA_WITH_CAMELLIA_256_GCM_SHA384",
    "TLS_RSA_WITH_CAMELLIA_256_CBC_SHA256",
    "TLS_RSA_WITH_CAMELLIA_256_CBC_SHA",
    "TLS_ECDH_RSA_WITH_CAMELLIA_256_GCM_SHA384",
    "TLS_ECDH_RSA_WITH_CAMELLIA_256_CBC_SHA384",
    "TLS_RSA_WITH_AES_128_GCM_SHA256",
    "TLS_RSA_WITH_AES_128_CCM",
    "TLS_RSA_WITH_AES_128_CBC_SHA256",
    "TLS_RSA_WITH_AES_128_CBC_SHA",
    "TLS_ECDH_RSA_WITH_AES_128_GCM_SHA256",
    "TLS_ECDH_RSA_WITH_AES_128_CBC_SHA256",
    "TLS_ECDH_RSA_WITH_AES_128_CBC_SHA",
    "TLS_RSA_WITH_AES_128_CCM_8",
    "TLS_RSA_WITH_CAMELLIA_128_GCM_SHA256",
    "TLS_RSA_WITH_CAMELLIA_128_CBC_SHA256",
    "TLS_RSA_WITH_CAMELLIA_128_CBC_SHA",
    "TLS_ECDH_RSA_WITH_CAMELLIA_128_GCM_SHA256",
    "TLS_ECDH_RSA_WITH_CAMELLIA_128_CBC_SHA256",
    "TLS_RSA_WITH_3DES_EDE_CBC_SHA",
    "TLS_ECDH_RSA_WITH_3DES_EDE_CBC_SHA",
    "TLS_EMPTY_RENEGOTIATION_INFO_SCSV"
  ],
  "ephemeral_keys_supported": true,
  "session_ticket_supported": true,
  "tls_compression_supported": false,
  "unknown_cipher_suite_supported": false,
  "beast_vuln": false,
  "able_to_detect_n_minus_one_splitting": false,
  "insecure_cipher_suites": {},
  "tls_version": "TLS 1.2",
  "rating": "Probably Okay"
}
```

> can we add some configuration tool in cloud builder?

Sure, I'll think about that once the current `dev` is also on `master` (don't wanna maintain two "feature sets") and once #1700 has been resolved.

marcelstoer, 2017-01-21:
Fixed with 88938b4.

commit 88938b44 (2017-01-21): Document mbed TLS cipher suites better, fixes #1749
closed 2017-01-21 by marcelstoer