nodemcu/nodemcu-firmware #1751: List Crontab | request feature

whoo, 2017-01-21:
### List all crontab scheduled

Ability to review already scheduled tasks, and get information, with functions like :
* cron.list() 
* cron:getschedule(x)
* ent:getschedule()

### Workarounds,
No clue for now, restart and set my scheduled tasks

Regards,
Dominique

djphoenix, 2017-01-21:
Hi there.
So, there is some issues with it.
Now, schedule string description (like `0 1 * * *`) is being parsed once you call `:schedule` method, and stored in packed bitmask format. There is difficult thing to process it back to string.
So list of entries without something that describing them is totally unusable (correct me if I wrong).

As module initiator and maintainer I prefer to keep it simple as possible, but if we can make it better - this is good :)

whoo, 2017-01-21:
I understand the process.
The best way could be sotring parameters on an external file.
(to keep persistence, and to simplify management).

tks

marcelstoer, 2017-01-21:
> So list of entries without something that describing them is totally unusable (correct me if I wrong).

ACK, scheduling tasks through `cron.schedule()` and then asking the module to give you a report of all tasks later ain't feasible IMO. It could be an option to invert this, though: describe frequency and task in a file and pass that to the module. Sounds familiar, I know.

> As module initiator and maintainer I prefer to keep it simple as possible

👍  That's a wise approach.

> The best way could be sotring parameters on an external file.

Nope, wouldn't like that. If you need to keep track of your tasks I suggest you do that manually in whatever form is most suitable to you.

closed 2017-01-23 by marcelstoer