nodemcu/nodemcu-firmware #1758: Traceroute

wolfgangr, 2017-01-23:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
Traceroute functionality

### Justification
Traceroute is one of the standard but more elaborated network diagnostic tools.

Well, I do not (yet) need it, but I accidentially found some point to begin with, so I tought this would be a good place to keep a reference. 

People were thinking about "sensor meshing" when I proposed an ethernet driver. So may be, in some future heavily meshed environment,  where cascaded subnets of only NODEMCUs are installed, sbdy may like to have a traceroute at hand.

### Workarounds
We have a ping now ( #1555 ) , which may suffice in most of the cases.
In many diagnostic setups, there will be some laptop or so available anywhere, which might provide tracroute functionality.

8<------------------------ END FEATURE REQUEST --------------------------------

### Where to start
Just by accident I realized, that in "SMCL"s github, where I sourced the ping function (#1555), threre is a [commit](https://github.com/smcl/nodemcu-firmware/blob/43e7dc9f70bb77c6598832606e9d940b14a60e44/app/modules/net.c) labelled **"initial attempt at traceroute".**

It might provide a good point to start with.
However, as discussed in #1555, I would not put into net.c, but into net_info.c, where I have put the ping.

@smcl or anybody picking up his code:
I'd point out  to my [comment](https://github.com/nodemcu/nodemcu-firmware/issues/1555#issuecomment-274182603) in the ping case regarding interface into net.c without braking backward compatibility.

smcl, 2017-01-23:
Hey - I'd be thrilled to give this a go. I only stopped short of implementing it because it would've ended up broadly the same as my ping example, just with a few tweaks, and I wanted to keep my little blog post simple.

wolfgangr, 2017-01-23:
Great :-)))

closed 2019-06-08 by TerryE