nodemcu/nodemcu-firmware #1759: Unable to initialize mqtt client

lenghonglin, 2017-01-24:
Hello,

I am running the lastest bin and after connecting to wireless I am unable to initialize the MQTT client

Here's what I get

m = mqtt.Client("clientid", 120, "user", "password")
stdin:1: attempt to call field 'Client' (a nil value)
I also tried
m=mqtt.Client()
stdin:1: attempt to call field 'Client' (a nil value)

flash new fireware still display this error。now i have no idea

FrankX0, 2017-01-24:
Apparently you are using a build which does not include the MQTT module.
Please generate a custom build with MQTT enable using https://nodemcu-build.com/

lenghonglin, 2017-01-25:
Nope,i built the newest include mqtt module .
	modules: file,gpio,mqtt,net,node,pwm,tmr,uart,wifi
 build 	built on: 2017-01-24 14:43
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

marcelstoer, 2017-01-25:
I just confirmed this ain't a bug with the firmware, works for me. Please use one of our support outlets, see #1010.

closed 2017-01-25 by marcelstoer