nodemcu/nodemcu-firmware #1760: Unable to load tls module

Lir10, 2017-01-25:
I've built latest nodemcu-firmware (Master branch) with SSL/TLS support . 
But when i run : 

`srv = tls.createConnection()`
i get : `attempt to index global 'tls' (a nil value)`

I've built the firmware from the cloud.

Thanks

Lir10, 2017-01-25:
I can confirm by the mail i got that tls has been added to the build: 

> This was built against the master branch and includes the following modules: cjson, crypto, encoder, file, gpio, mqtt, net, node, tmr, uart, wifi, **tls**

Lir10, 2017-01-25:
Looks like it's working only on the dev branch

closed 2017-01-25 by Lir10

djphoenix, 2017-01-25:
@marcelstoer is there a bug in cloud builder?

marcelstoer, 2017-01-25:
It's notoriously difficult to prove the absence of bugs 😉 

However, this ain't a bug. The TLS module will only be available on `master` after the [next `dev`-to-`master` merge](https://github.com/nodemcu/nodemcu-firmware/milestone/4).

Lir10, 2017-01-25:
@marcelstoer Thanks for lettings us know :)

@djphoenix Few questions:
A. Is it possible to connect to tls server from esp8266 with this tls module using client certificate  ?

B. In the docs it says something about saving the certificate inside 

> tls.cert.verify()

, is it possible to store  all the certificate chain ?

marcelstoer, 2017-01-25:
@Lir10 our documentation is bound to a specific branch since we keep it in the same repo as the code, there's a difference between http://nodemcu.readthedocs.io/en/dev/en/modules/net/ and http://nodemcu.readthedocs.io/en/master/en/modules/net/

Lir10, 2017-01-25:
@marcelstoer As i can see from the docs (dev branch) `net.cert` Module has gone to the TLS module, but it's not clear from tls docs how can we add all the certificate chain to tls.cert.verify() as a string.

djphoenix, 2017-01-25:
@Lir10, I totally not understand how (and why) tls module works right now. This is a remaining part of my `net` module rework (#1379). And we will completely rewrite it sooner or later (#1700).

About certificate verification - it goes to old `espconn` layer, and, so, passed through it to new mbed TLS layer (since 2.0.0 - #1435). `espconn` is a mess at all...

Lir10, 2017-01-25:
@djphoenix What do you suggest to use for the meantime for secured connection untill tls module will be completed (or it's not possible for now to create secured connection with the current stable version) ?

As for now , it's not clear to me how to create simple client ssl connection even without client certificate

djphoenix, 2017-01-25:
As in current `dev` docs, you can use both old-style [`net.createConnection(net.TCP, 1)`](http://nodemcu.readthedocs.io/en/dev/en/modules/net/#netcreateconnection) and new-style [`tls.createConnection()`](http://nodemcu.readthedocs.io/en/dev/en/modules/tls/#tlscreateconnection) API. Please note that old-style API is kept for backward compatibility and will be deprecated.

So after creating connection you may [`:connect`](http://nodemcu.readthedocs.io/en/dev/en/modules/tls/#tlssocketconnect) it to remote server. And it works.
I never used certificate validation with nodeMCU (old SSL or even mbedTLS), but you can see [example](http://nodemcu.readthedocs.io/en/dev/en/modules/tls/#tlscertverify) of passing your root CA cert to nodeMCU. I think it should work.

Lir10, 2017-01-26:
@djphoenix Thanks , will try to test it with CA cert only. 
It would be great if we would be able to add client certificate to tls module api in the near future !

tcpipchip, 2021-08-04:
hi, how to enable the TSL module on LUA BUILD (ESP32)
attempt to index global 'tls' (a nil value)
Thanks!