nodemcu/nodemcu-firmware #1762: Old-style net API deprecation

djphoenix, 2017-01-26:
Now we have new APIs for creation TLS connections and UDP sockets, and so old APIs that kept only for backward compatibility (just convenience calls):

- `net.createConnection(net.UDP)` -> `net.createUDPSocket()`
- `net.createServer(net.UDP)` -> `net.createUDPSocket()`
- `net.createConnection(net.TCP, 1)` -> `tls.createConnection()`

I propose to:

1. After upcoming master drop, set `net.createConnection()` behaviour to create **plain tcp connection** by default (without options).
2. Add deprecation message to docs and to firmware (#1538).
3. After (upcoming+1) master drop, completely remove deprecated feature, and keep only separated APIs.

/cc @marcelstoer @jmattsson @devsaurus comments?

marcelstoer, 2017-01-26:
I basically agree but this

> Add deprecation message to docs

should be done _before_ the upcoming merge.

jmattsson, 2017-01-26:
No objections from me.

devsaurus, 2017-01-26:
For sure the way to go :+1:

djphoenix, 2017-01-26:
@marcelstoer 
> I basically agree but this [deprecation message to docs] should be done before the upcoming merge.

Can you add right message by yourself? I have not excellent english, you see :)

marcelstoer, 2017-01-26:
@djphoenix issue number 2 from your list above can be ticked off, see http://nodemcu.readthedocs.io/en/latest/en/modules/net/#netcreateconnection

marcelstoer, 2017-02-11:
> Add deprecation message to the firmware (requires #1538).

That PR is now ready.

marcelstoer, 2019-01-12:
This is fast reaching its 2nd birthday. @nwf retired the static timers in #2603. Maybe it would be a good time to finally close this one here as well.

nwf, 2020-01-07:
Hey, more dead code to eliminate.  I'll  take this on, too, as part of my rampaging around the network stack.

marcelstoer, 2020-01-07:
> more dead code

And dead _documentation_, too 😉

HHHartmann, 2021-01-02:
I can`t see anything to do here anymore. Documentation looks good and the implementation also does not allow for options anymore.

Can we close this?

commit d2fbf40c (2017-01-26): Add UDP-related deprecation notes, partially fixes #1762
PR #1844 (2017-03-07): Deprecation messages for convenient net.create(Server/Connection) calls
Fixes #1762 (pt. 3), also add default parameter values in docs.
closed 2021-01-02 by nwf