nodemcu/nodemcu-firmware #1764: loadlib style dynamic linking from lua land into C code

wolfgangr, 2017-01-26:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
lua loadlib, as described by chapter [8.2 – C Packages](http://www.lua.org/pil/8.2.html) of the "Programming in Lua" book, or comparable functionality to dynamically call arbitrary C code in the firmware from Lua land

### Justification
- Increasing shift of module interfacing from C to lua
- maybe dropping the requirement for C interfacing for many module situations
- maybe drastically reducing the number of C modules required
- maybe drastically reducing the maintenance effort for nodemcu firmware in the long run

### Workarounds
The way it is done now:
- Expose any C function and its associated data in a more or less thin layer to lua land
- recompile/reflash for additional modules
- [cloud build service](https://nodemcu-build.com/)
- have to keep C and lua code in sync
- do more stuff in C than conceptually reqired to reduce lua mainenance burden
- clobbered repository of lua land code base

8<------------------------ END FEATURE REQUEST --------------------------------

wolfgangr, 2017-01-27:
While trying to get my first nodemecu PR flying, I had learned that it were preferrable to move as much code as possible from the 'C side' over the 'interface river' into 'lua land':
https://github.com/nodemcu/nodemcu-firmware/pull/1757#issuecomment-275061025

While reading the famous PIL "Programming in Lua" book as quoted above, I encountered the concept of calling right into dynamically linkable libraries:
> "Lua provides all the functionality of dynamic linking in a single function, called `loadlib`.
 It has two string arguments: the complete **path of the library** and the **name of an initialization function**. "

I also see that there is already a `struct` module available in `nodemcu`, able to convert back and forth from C data types.

So my idea was if it were possible to skip the C interface layer completely and do all the interfacing from the lua land side alone, right into the middle of C production code already present.

As a first disappointment, when I try to call `loadlib` on my ESPlorer console, I get
```
=package.loadlib("ping.o","ping_by_hostname") 
nil	dynamic libraries not enabled; check your Lua installation	absent
```
which we can trace back to `/app/lua/loadlib.c`
```
#else
** {======================================================
** Fallback for other systems
** =======================================================
   ....
#define DLMSG	"dynamic libraries not enabled; check your Lua installation"
```
I see the obvious reason for omitting a "normal" loadlib implementation:  
**To save scarce flash footprint, the symbol tables are dropped** from the binary before flashing. 

So either we enable some of them kept again, but that still reqires decisions about which interfaces to present and which not. This way, we do not really win much, compared to the present situation - neither in terms of memory nor in terms of manageability.

The alternative might be to provide the symbol information in a pathway bypassing the firmware flash bottleneck.
The obvious way I think would be a serialized **lua table, extracted from the build tree**, stripped down by proper **selection criteria**, loaded into the file system like any other lua code.

This is the final result as achieved by the current mechanism, as we had to resemble it:
```
=net_info.ping
ASSERT@lapi.c(417): (((o)->tt) == 4)
lightfunction: 0x4026099c
```

So let's go searching....
```
...nodemcu-firmware$ grep -nri "net_info.ping" .
Binary file ./app/modules/.output/eagle/debug/obj/net_info.o matches
Binary file ./app/modules/.output/eagle/debug/lib/libmodules.a matches
./app/modules/net_info.c:111:static int net_info_ping(lua_State *L)
./app/modules/net_info.c:167:  { LSTRKEY( "ping" ),             LFUNCVAL( net_info_ping ) },
Binary file ./app/.output/eagle/debug/image/eagle.app.v6.out matches
./app/mapfile:3269: .literal.net_info_ping
./app/mapfile:9801: .text.net_info_ping
Binary file ./bin/0x10000.bin matches
```

I peeked into the `mapfile`, but could not match the adresses there.
Presumably I haven't yet understood the syntax and offset there.

However, I could extract the correct address with the help of **gdb**:
```
...$ xtensa-lx106-elf-gdb app/.output/eagle/debug/image/eagle.app.v6.out
     .....
(gdb) disass net_info_ping
Dump of assembler code for function net_info_ping:
   0x4026099c <+0>:     addi    a1, a1, -32
```
As we can see, `net_info_ping` is mapped precisely to the same address `0x4026099c` as above.
I could not manage yet to call this adress as a function - lua still considers it a number.  But I still have some chapters of PIL left to read. And if nothing else helps, I think a 3 line C for casting an adress on the stack into a lua function shold do the job.

I think the harder part will be to keep data structures consistent. While the nodemcu struct module might offer the general mechanism for that, keeping data definitions in sync may still provide some challenge.

I just encountered the lua object called registry, which shows some trace of `LOADLIB` labels, but I could not yet figure out any details of their meaning:
```
> for k,v in pairs(debug.getregistry()) do
print (k,v)
end
tmr.timer	romtable: 0x4026f2d8
_LOADED	table: 0x3fff07a8
net.tcpsocket	romtable: 0x4026eda8
INT	romtable: 0x40270330
_LOADLIB	romtable: 0x40270760
LOADLIB: ping.o	userdata: 0x3ffeff68
net.udpsocket	romtable: 0x4026ecd0
LOADLIB: bar	userdata: 0x3ffeff90
net.tcpserver	romtable: 0x4026eec8
```

I look forward to comments in all directions

Qestions as they come to my mind:
- is this possible at all?
- or are there any irremovable obstacles?
- is it desirable at all?
- is there maybe already a simple solution at hand?
- is there any code we might reuse or build upon?

TerryE, 2017-01-28:
@wolfgangr, this list is primarily for those collaborating on developing the NodeMCU core or the library externsions. This sort of "how do I" or "wouldn't it be nice to" topic would be best discussed elsewhere as per our [Support](https://nodemcu.readthedocs.io/en/dev/en/support/) page. 

The Lua dynamic library load feature is layered on the underlying OS's own dynamic load functionality, except in this case the Espressif non-OS SDK doesn't offer this functionality, since the entire build is stored in on-module Flash.  Sorry.

closed 2017-01-28 by TerryE