nodemcu/nodemcu-firmware #1770: Wifi speed too low

saoodmohd, 2017-01-29:
Hello, I am working with ESP8266-12 with nodemcu firmware and trying to send data to my local server using TCP protocol. I wanted to test the wifi transfer rate with some test data and found speed to be too low (as low as 10KBps). The following is the code

```
print(tmr.time());
data='123456789_123456789_123456789_123456789_123456789_123456789_123456789_123456789_123456789_123456789_';
length=1000;
check=(5+length*string.len(data));
conn=net.createConnection(net.TCP, 0);
conn:on("receive",  function(conn2,pl) 
        print(pl);
        print(tmr.time());
        conn2:close();
        conn=nil;
        collectgarbage();
end);
str="POST /ml/index.php HTTP/1.1\r\nHost: "..serverip.."\r\nConnection: close\r\nContent-Length: "..check.."\r\nContent-Type: application/x-www-form-urlencoded\r\nAccept: */*\r\n\r\ndata=";
conn:on("sent",function(conn,c) 
	if(length>0)then
		conn:send(data)
		length=length-1
	end
end);
conn:on("connection", function(conn,c)
	print('Conneted!')
	conn:send(str);
end);
conn:connect(port,serverip);
```
With this method I am getting bout 8-10KBps. Which is too low. Or Is there something wrong about the method I am using?

marcelstoer, 2017-01-29:
See #1010.

closed 2017-01-29 by marcelstoer

pjsg, 2017-01-29:
What firmware version are you using?

saoodmohd, 2017-02-01:
I am using the custom built firmware from https://nodemcu-build.com/ 
I hope its not too old as i built it recently. Following is what it says at statup
```
NodeMCU custom build by frightanic.com
	branch: master
	commit: 81ec3665cb5fe68eb8596612485cc206b65659c9
	SSL: false
	modules: cjson,file,gpio,http,mqtt,net,node,tmr,uart,wifi
 build 	built on: 2017-01-29 10:26
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
```

pjsg, 2017-02-01:
Can you try the `dev` build. The networking function were rewritten and things *may* have gotten better. The issue with `master` is that the `sent` callback is not invoked until the data packet has actually reached the server. I'm not sure if the new code does some local buffering or not.

saoodmohd, 2017-02-02:
I tried the dev built but still no improvement on speed. So this time I tried to send the data all at once.
```
data='123456789_123456789_123456789_123456789_123456789_123456789_123456789_123456789_123456789_123456789_'
check=(5+length*string.len(data));
conn=net.createConnection(net.TCP, 0);
conn:on("receive",  function(conn2,pl) 
	print(pl);
	print(tmr.time());
	conn2:close();
	conn=nil;
	collectgarbage();
end);
str="POST /ml/index.php HTTP/1.1\r\nHost: "..serverip.."\r\nConnection: close\r\nContent-Length: "..check.."\r\nContent-Type: application/x-www-form-urlencoded\r\nAccept: */*\r\n\r\ndata=";
	
conn:on("connection", function(conn,c)
	print('Conneted!')
	conn:send(str);
	for i =1,length do
		conn:send(data)
	end
end)
conn:connect(port,serverip);
```
But now the send function is not able to handle the huge data and it says
PANIC: unprotected error in call to Lua API (init.lua:56: out of memory)
line 56 is  'conn:send(data)'