nodemcu/nodemcu-firmware #1775: Cant send messages to connected client after WiFi connection has been established

Lir10, 2017-02-02:
Before i connect esp to wifi network i can successfully send messages to connected client ( port 8080) , but after i connect to a wifi network the client doesnt receives anything.
```
wifi.setmode(wifi.STATIONAP)
sock=nil

wifi.eventmon.register(wifi.eventmon.STA_CONNECTED, function(T)
    sock:send("Hello World")
end)

function receiver(sck, data)
    t = cjson.decode(data)
    wifi.sta.config({ssid=t.ssid, pwd=t.pwd})
end

sv:listen(8080,function(conn)
    conn:on("receive", receiver)
end)
```

Lir10, 2017-02-02:
UPDATE: I've just tested it out and seems like when the esp tryies to connect wifi it diconnect the current AP socket and then activate it again. so i suppose the client should reconnect to the esp again after wifi connecation has been made?

devsaurus, 2017-02-02:
Did you consider the following implications of stationap mode?
http://bbs.espressif.com/viewtopic.php?t=324

Lir10, 2017-02-02:
Thanks , that can explain the problem i had.

So the conclusion is that the client should prepare to reconnect to the esp in softAP+STATION mode

closed 2017-02-05 by Lir10