nodemcu/nodemcu-firmware #1777: LWM2M for NodeMCU

joakimeriksson, 2017-02-03:
### Missing feature - LWM2M
LWM2M is currently not part of the NodeMCU software stack

### Justification 
LWM2M is becoming more and more common for IoT to cloud connectivity. Multiple commercial cloud providers support it and it is one of the (few) already mature standards.

### Workarounds
I have an initial Lua integration of our Contiki-based implementation of LWM2M for NodeMCU. If this is of interest to anyone but me I would be happy to share it. I have just managed to post sensor data (fake, but still) to the leshan.eclipse.org LWM2M server from my NodeMCU.

marcelstoer, 2017-02-11:
This is the first time I heard of LWM2M - always interesting to learn about new stuff. I briefly scanned https://en.wikipedia.org/wiki/OMA_LWM2M, http://www.slideshare.net/zdshelby/oma-lightweightm2-mtutorial and http://www.openmobilealliance.org/wp/Overviews/lightweightm2m_overview.html.

> I have an initial Lua integration of our Contiki-based implementation of LWM2M for NodeMCU. If this is of interest to anyone but me I would be happy to share it.

Sure, we love PRs 😉 I guess you know about https://github.com/nodemcu/nodemcu-firmware/blob/master/CONTRIBUTING.md, https://github.com/nodemcu/nodemcu-firmware/wiki/%5BDRAFT%5D-How-to-write-a-C-module and http://nodemcu.readthedocs.io/en/latest/en/extn-developer-faq/

joakimeriksson, 2017-02-11:
Sounds good. Then I will do some improvements on the Lua integration, prepare some documentation and later prepare a PR for the LWM2M support! I'll get back as soon as I am getting there!

marcelstoer, 2017-04-30:
Feel free to come back with a PR once you're ready - closing for now.

closed 2017-04-30 by marcelstoer

joakimeriksson, 2017-05-01:
I will - thanks.