nodemcu/nodemcu-firmware #1778: Persistent tcp connection

Lir10, 2017-02-05:
Hi,

I'm wondering whether it's possible or even efficient to create persistent tcp connection using the net module, or is it better to create a connection every time i need to send a message to the server? 

code example : 

```
conn = net.createConnection(net.TCP, 0)
  conn:on("connection", function()
    print("connected to server")
  end)
  conn:on("disconnection", function()
    print("disconnected from server")
  end)
  conn:connect(port,host)
```

FrankX0, 2017-02-08:
By experience I know that a persistent tcp connection works.
Only now and then I experience that the connection is closed and has to be opened again, which can be done in the "disconnection"  callback.

Lir10, 2017-02-10:
@FrankX0  Thanks gave it a shot and it works for the moment flawlessly!

closed 2017-02-10 by Lir10